flipperdevices/qFlipper #30: A few propositions about the project structure

ValentiWorkLearning, 2021-12-09:
I've looked through the QML and C++ structure of the project. Also, I've checked the building procedure. I have a few suggestions which can be useful. Also, I can try to implement them in the fork.

1. Migrate to Qt 6.2. The QML structure can be simplified according to the major changes in the core.
2. Rewrite building procedure to CMake. I'm familiar with this and have experience of using Qt/QML with the CMake integration. Also, it simplifies cross-platform building and even cross-architecture compilation
3. Change C++ standard to C++17 since all of the compilers support it normally.

What do you think about these ideas?

aprosvetova, 2021-12-10:
Qt 6.2 might also be a step towards Apple Silicon supported universal macOS build 😏

gsurkov, 2021-12-10:
First and foremost, thank you for your suggestions!

As of now, I cannot make an informed decision before knowing the following:
1. What exact advantages will the move to Qt 6.2 bring, compared to 5.15? Will the there be any regressions? Last time I checked, the Qt 6.x branch lacked several important modules.
2. Again, what are the benefits of using CMake instead of qmake? Qmake comes pre-installed with every Qt distribution and is closely integrated with the framework itself and QtCreator as well. The only real disadvantage of the latter I can see is that it is not trivial to integrate with third-party projects, since they rarely use qmake themselves.
3. Switching to a newer C++ standard might be a good way to future-proof the code, but I can't imagine any of its features that would benefit the project immediately (howeve, I'm open to changing my mind on this point if presented with good arguments). More so, we don't want to force people to using the latest-and-greatest compilers. For example, the Linux build (which is, frankly, the most problematic) uses Ubuntu 18.04, as it has proved to be still widely used. 

I would really like to see some answers to the questions above as I also would not want to be stuck in my current ways and see how things could be done better.

> Qt 6.2 might also be a step towards Apple Silicon supported universal macOS build

Yes, this is a very good argument in favour of the transition.

ValentiWorkLearning, 2021-12-10:
@gsurkov 
Thanks for your answer.
One of the major points of migration to Qt 6 - simplifying QML files. Take a look at: 
https://wiki.qt.io/New_Features_in_Qt_6.0
```
Version numbers are optional in the QML language. An import without any version number imports the latest version of the module. An import with only a major version number imports the module with specified major version and the latest minor version.
```
 At least, you can be able to get rid off the versions of the imported QML files.

2. CMake points:
- With CMake you are able to quickly integrate any 3rd party to the application if this library supports CMake
- With CMake it's easy to use Conan/vcpkg pacakge manager for the dependencies
- With CMake you are free from any IDE. On Windows Qt Creator debugger lacks of features, in compare to Visual Sudio + Qt plugin for VS development. I can show you some screens of the debugging Qt/QML application from VS. It's quite closer to the gdb facilities.  You can even open the QObject structure to observe the internals of QObject and obtain the pretty-printed QString in the Watch window.
3. Newer C++ standard. Almost all of the major distributions are supplied with at least gcc 8.3.0 which supports C++17 standard. A short list of benefits:
- Class template argument deduction
- if constexpr
- std::filesystem

gsurkov, 2021-12-12:
Thank you for providing the answers @ValentiWorkLearning
After considering all of the above and doing some additional research I have nothing against your suggestions, so you may go ahead and implement them if you like.

There are a couple of requirements for your changes to be accepted, though.
- New build system must not break the existing CI workflow (minor adjustments are acceptable).
- The minimum required version of CMake must be no higher than one that comes with the official Qt for Windows installer.
- The AppImage executable for Linux must run on Ubuntu 18.04 or any other distro with an equal or greater libc version.

If you will have any questions, please feel free to ask.

closed 2021-12-12 by gsurkov