flipperdevices/qFlipper #41: dfu-util instructions incomplete?

abliss, 2022-03-11:
I followed the dfu-util instructions since I can't seem to run the Qt app. It seems to successfully flash my firmware but does not provide a radio stack or SDCard database, so the flipper is not fully usable. Could you please add some instructions for installing these from a commandline?

gsurkov, 2022-03-12:
Could you please describe why you weren't able to run the application?
You may use [this description](https://github.com/flipperdevices/qFlipper/blob/dev/BUGS.md) as a general guideline. Please include your OS type and version as well.

zhovner, 2022-03-12:
Maybe you can run command line version of qFlipper?

abliss, 2022-03-12:
can the cli update the sdcard database?

gsurkov, 2022-03-13:
@abliss yes, it does everything the graphical application does. If you wish giving it a try, refer to [this documentation](https://github.com/flipperdevices/qFlipper/blob/dev/tool/README.md).

abliss, 2022-03-13:
Thanks, I had indeed looked at that doc, but I can't figure out which command creates the sdcard database. Is it `firmware`?

gsurkov, 2022-03-14:
@abliss run it without any arguments, it will update the firmware and install the databases.