flipperdevices/qFlipper #42: qFlipper displays "Invalid Device", unable to update firmware

tlhunter, 2022-03-12:
I just received my flipper in the mail. It has a full charge and an older version of firmware, I think 0.43 or 0.40, something from December. I successfully scanned a few cards and IR remotes then discovered the internal storage was too low to scan more things. I inserted a 32GB micro SD card and formatted it using Flipper.

At this point I decided to update the firmware.

I'm using the Linux AppImage binary:

```sh
./qFlipper-x86_64-0.9.0.AppImage
```

The GUI runs but it is unable to connect to my device:

[image]

I'm using a laptop with no peripherals connected but the error persists.

Following [the docs](https://docs.flipperzero.one/8PvC5Nim38M7V0_khYFNp), I held down back and OK for thirty seconds until the device rebooted. The GUI still does not allow me to connect to the Flipper. `lsusb` confirms that the OS sees the device and that it is now in recovery mode:

```
Bus 006 Device 023: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
```

Is there a better way to update the firmware? Perhaps writing a file to the sd card?

tlhunter, 2022-03-12:
Here are some additional logs when plugging/unplugging the device from the computer with qFlipper running:

```
[APP] qFlipper version 0.9.0 commit 62c4653c 2022-03-04T08:08:03
[APP] OS info: Ubuntu 20.04.4 LTS 20.04 5.15.23-051523-generic
[DEBUG] Failed to open device
[DEVREG] Invalid device detected: VID_0x0:PID_0x0, ignoring it
[UPDATES] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
[UPDATES] Fetched update information from https://update.flipperzero.one/firmware/directory.json
[BACKEND] Finalized current operation
[DEBUG] Failed to open device
[DEVREG] Invalid device detected: VID_0x0:PID_0x0, ignoring it
```

Rebooting the device doesn't help. Switching cables doesn't help either.

tlhunter, 2022-03-12:
I did the unthinkable and used my Windows desktop to install the firmware update. It worked without a hitch. I'll keep this thread open since there seems to be an issue with the AppImage build.

zhovner, 2022-03-12:
Thanks. Your distro is `Ubuntu 20.04.4 LTS 20.04 5.15.23-051523-generic`?
What about user permissions to access TTY devices?

gsurkov, 2022-03-12:
It does look like the device access rules have not been set up.
Please run [this script](https://github.com/flipperdevices/qFlipper/blob/dev/setup_rules.sh) and try again.

zhovner, 2022-03-12:
@gsurkov what do you think about .deb repository and adding this script to install process? Maybe we can find someone who can maintain it.

gsurkov, 2022-03-12:
@zhovner Ideally, we should have native packages for all popular linux distributions. And indeed, we need maintainers.

benklop, 2022-03-13:
I just encountered this issue as well, on Manjaro. the linked script did the trick.

gsurkov, 2022-03-17:
Made sure to display a message about access rules when this happens. Closing for now.

closed 2022-03-17 by gsurkov

tlhunter, 2022-03-18:
New message is very helpful. The linked script worked like a charm.

acheong08, 2023-10-10:
> Please run [this script](https://github.com/flipperdevices/qFlipper/blob/dev/setup_rules.sh) and try again.

Fixed it!

nekroun, 2024-01-18:
> It does look like the device access rules have not been set up. Please run [this script](https://github.com/flipperdevices/qFlipper/blob/dev/setup_rules.sh) and try again.

This was it for me, thank you so much!

Martin-105, 2024-06-27:
It didn't help me, it gives me an error: zsh: event not found: /bin/sh