flipperdevices/qFlipper #43: Failed to build: "Nothing to be done with 'qmake all'."

Lucaslhm, 2022-03-12:
While trying to build the file-manager branch on OSX Big Sur 11.6.4 I recieved the following:

```
lucas@Lucass-MBP qFlipper % ./build_mac.sh     
++ pwd
+ PROJECT_DIR=/Users/lucas/Desktop/qFlipper
+ PROJECT=qFlipper
+ BUILD_DIRECTORY=build_mac
+ [[ -d build_mac ]]
+ rm -rf build_mac
+ mkdir build_mac
+ cd build_mac
+ qmake -spec macx-clang CONFIG+=release CONFIG+=x86_64 -o Makefile ../qFlipper.pro
+ make qmake_all
cd 3rdparty/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/3rdparty/3rdparty.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
WARNING: Failure to find: nanopb/pb_common.c
WARNING: Failure to find: nanopb/pb_decode.c
WARNING: Failure to find: nanopb/pb_encode.c
cd 3rdparty/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
make[1]: Nothing to be done for `qmake_all'.
cd dfu/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/dfu/dfu.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
cd dfu/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
make[1]: Nothing to be done for `qmake_all'.
cd plugins/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/plugins/plugins.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
cd plugins/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
cd flipperproto0/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/plugins/flipperproto0/flipperproto0.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
cd flipperproto0/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
make[2]: Nothing to be done for `qmake_all'.
cd protobufinterface/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/plugins/protobufinterface/protobufinterface.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
cd protobufinterface/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
Some of the required modules (false) are not available.
Skipped.
cd backend/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/backend/backend.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
cd backend/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
make[1]: Nothing to be done for `qmake_all'.
cd application/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/application/application.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
Project WARNING: Qt has only been tested with version 10.15 of the platform SDK, you're using 12.1.
Project WARNING: This is an unsupported configuration. You may experience build issues, and by using
Project WARNING: the 12.1 SDK you are opting in to new features that Qt has not been prepared for.
Project WARNING: Please downgrade the SDK you use to build your app to version 10.15, or configure
Project WARNING: with CONFIG+=sdk_no_version_check when running qmake to silence this warning.
cd application/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
make[1]: Nothing to be done for `qmake_all'.
cd tool/ && /usr/local/bin/qmake -o Makefile /Users/lucas/Desktop/qFlipper/tool/tool.pro -spec macx-clang CONFIG+=release CONFIG+=x86_64
Project WARNING: Qt has only been tested with version 10.15 of the platform SDK, you're using 12.1.
Project WARNING: This is an unsupported configuration. You may experience build issues, and by using
Project WARNING: the 12.1 SDK you are opting in to new features that Qt has not been prepared for.
Project WARNING: Please downgrade the SDK you use to build your app to version 10.15, or configure
Project WARNING: with CONFIG+=sdk_no_version_check when running qmake to silence this warning.
cd tool/ && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile qmake_all
make[1]: Nothing to be done for `qmake_all'.
+ make -j9
make[1]: *** No rule to make target `nanopb/pb_common.c', needed by `pb_common.o'.  Stop.
make: *** [sub-3rdparty-make_first] Error 2
make: *** Waiting for unfinished jobs....
+ macdeployqt qFlipper.app -executable=qFlipper.app/Contents/MacOS/qFlipperTool -qmldir=/Users/lucas/Desktop/qFlipper/Application -verbose=1
Error: Could not find app bundle "qFlipper.app"
```

I ran `brew install qt5` to install Qt >= 5.15.0

The notable error here that I see is `make[1]: Nothing to be done for qmake_all.`

gsurkov, 2022-03-12:
I can see the following possible issues:
1. Submodules might not have been cloned. Run `git submodule init && git submodule update` to do so.
2. The platform SDK is too new. Install the recommended SDK version as in the warnings. This might not be the immediate source of the problem, but can cause issues later on.
3. Have you linked qt5 as the default qt version? run `brew link qt5 --force` if you haven't done it yet.

Lucaslhm, 2022-03-13:
The issue was in fact that I did not clone the submodules.

closed 2022-03-13 by Lucaslhm