flipperdevices/qFlipper #44: Remember installer choices for updates

alexwaibel, 2022-03-16:
During the qFlipper installation I've selected a different install path as well as opted out of the desktop icon. However, when I go to update flipper, the installer has forgotten these choices. It would be helpful to save these choices somewhere and use them as the defaults for the update installer.

gsurkov, 2022-03-17:
Thanks for the suggestion! Will look into it later, when more critical issues will have been fixed (unless someone makes a PR implementing it).

zhovner, 2022-04-04:
Fixed in this branch, please test https://update.flipperzero.one/builds/qFlipper/nsis_modern_ui/

gsurkov, 2022-04-06:
Is now available in the [official release binary](https://flipperzero.one/update).

closed 2022-04-06 by zhovner