flipperdevices/qFlipper #46: [Feature Request] Text editor in file manager

fennectech, 2022-03-17:
ive been trying the flipper zero out and absolutely love it   ive also got a build of qflipper with the file manager but id love to be able to edit files directly from qflipper   ive done a good bit of playing arround with .ir and .nfc files and found them dead simple to modify with a text editor   having one in qflipper would streamline editing and be quite nice to have

gsurkov, 2022-03-18:
It seems a bit out of qFlipper's scope, but I will discuss whether it should be implemented with the team.

zhovner, 2022-03-23:
We can't compete with the editors like `Sublime Text` that have a code highlight and more so better not even try.

fennectech, 2022-03-23:
It doesnt have to be anything like that. Just a simple way to edit text files without pulling them off the device.   Something like notepad

zhovner, 2022-03-24:
Thanks, will keep your wish in mind while planning a new roadmap.

closed 2022-03-24 by zhovner