flipperdevices/qFlipper #49: Missing protobuf files

reinerh, 2022-03-26:
I noticed that the repository contains generated files in `plugins/flipperproto0/messages/`, that have been generated with nanopb. But the protobuf files for actually regenerating the files are missing.

Please include the source for the generated files, and preferrably regenerate them also during build instead of using the pre-generated ones.

zhovner, 2022-03-26:
Did you run this?

```
git submodule init
git submodule update
```

reinerh, 2022-03-26:
This only makes the 3rdparty submodules (nanopb, driver-tool) available, doesn't it?
I'm not looking for nanopb itself, it is already available in my distribution. But for the files that were used as input to generate e.g. `plugins/flipperproto0/messages/application.pb.h`

DrZlo13, 2022-03-27:
We keep the protobuf sources in a separate repo, so just update submodules.

reinerh, 2022-03-27:
Thanks, I have found them now in a different repository of this organization: https://github.com/flipperdevices/flipperzero-protobuf
Though that repository is not part of the submodules:
https://github.com/flipperdevices/qFlipper/blob/977e756f890847e60d4456410b388d70d3da5f88/.gitmodules#L1-L6

closed 2022-03-27 by reinerh