flipperdevices/qFlipper #50: Firmware Update Should Be Blocked If No SD Card Installed

maximombro, 2022-03-27:
**Describe the bug**
Connected Flipper Zeros can attempt to be updated without an SD Card installed. The [SD Card Basics documentation](https://docs.flipperzero.one/basics/sd-card) explicitly states that "it is also needed to update the firmware, so it is important to install the SD card BEFORE updating the firmware."

**To Reproduce**
Steps to reproduce the behavior:
1. Have a functional Flipper Zero with **no** SD Card installed.
2. Start the qFlipper desktop application.
3. Plug in the Flipper Zero to the desktop with qFlipper open.
4. Notice that the currently attached Flipper Zero is identified as having no SD Card.
5. Notice you are allowed to start a firmware update despite lacking an SD Card.

**Expected behavior**
Firmware update is blocked until an SD Card is properly installed.

**Actual behavior**
Firmware update can be executed but cannot complete because of the lack of SD Card. If Firmware update is executed, the Flipper will become inoperable and must be fixed using the qFlipper's "Repair" functionality.

I have not yet been able to test if installing an SD Card and clicking the "Repair" button will actually repair the Flipper from this state.

**Additional troubleshooting**
- [1333 Line Full Log](https://pastebin.com/zjVrBqQ2)

**Additional context**
n/a

gsurkov, 2022-03-28:
Your problem is most likely not connected to the absence of the SD card. The log you have attached shows a problem during radio stack update (it is handled by Firmware Update Service(FUS) by ST microelectronics and can fail sometimes).

I would suggest to use the Repair feature and see if it manages to fix the firmware. Trying it on another computer may also help in case the problem persists (unfortunately, this update process is very dependent on client hardware).

As for allowing a firmware update without an SD card installed, it was a design choice and cannot cause any direct problems (such as bricking the device).

maximombro, 2022-03-28:
I was just able to attempt the "Repair" functionality _and_ an SD Card installed. However, if the statement in the [SD Card Basics documentation](https://docs.flipperzero.one/basics/sd-card) is incorrect it should be corrected to avoid future mistakes. Conversely, even if this needs to be turned into a "feature" request, the function of qFlipper should match the documentation associated with the shared product.

zhovner, 2022-04-06:
Thanks for your report @maximombro 
The SD card required for several features like Infrared and Sub-GHz and qFlipper installs databases on it. But general firmware update is supported without and SD card too.

We will closer look at documentation wording, thank you.

closed 2022-04-06 by zhovner