flipperdevices/qFlipper #52: Unable to fetch updates from the server (SSL Handshake failed)

NoraCodes, 2022-04-03:
**Describe the bug**
I was trying to update my Flipper from 0.50 to 0.53. When I opened qFlipper, I got the error message: "Cannot connect to update server."

**To Reproduce**
Steps to reproduce the behavior:
1. Open qFlipper
2. Observe error message

**Expected behavior**
qFlipper can contact the update server.

**Actual behavior**
qFlipper could not contact the update server.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? **Yes.**
3. Does reconnecting/rebooting Flipper solve it? **No.**

**Additional context**
I'm using the Ubuntu 22.04 beta, but I seriously doubt this is an issue with that.

I was able to update by manually downloading the firmware file.

```
[APP] qFlipper version 1.0.0 commit 69e0b6a7 2022-03-25T08:25:55
[APP] OS info: Ubuntu Jammy Jellyfish (development branch) 22.04 5.15.0-23-generic
[REG] Detected new device: VID_0x483:PID_0x5740
[qt.network.ssl] QSslSocket: cannot call unresolved function SSL_get_peer_certificate
[qt.network.ssl] QSslSocket: cannot call unresolved function SSL_get_peer_certificate
[UPD] Failed to fetch update information: Network error: SSL handshake failed
[UPD] Failed to fetch update information: Network error: SSL handshake failed
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) System Device Info START
[RPC] (1) System Device Info SUCCESS
[RPC] (2) Storage Info @/ext START
[RPC] (2) Storage Info @/ext SUCCESS
[RPC] (3) Storage Stat @/ext/Manifest START
[RPC] (3) Storage Stat @/ext/Manifest SUCCESS
[RPC] (4) System Get DateTime START
[DBG] Flipper time skew is -635 milliseconds
[RPC] (4) System Get DateTime SUCCESS
[RPC] (5) System Set DateTime START
[RPC] (5) System Set DateTime SUCCESS
[RPC] Stopping RPC session...
[RPC] Attempting to unload protobuf plugin...
[RPC] Unloaded protobuf plugin.
[RPC] RPC session stopped successfully.
[REG] Registering the device
[DEV] Version: 0.53.0 commit: 5f582abf radio: 1.13.0
[BKD] Current device changed to "[REDACTED]"
[RPC] Starting RPC session...
[RPC] RPC session started successfully.
[RPC] (1) Gui Start SreenStream START
[RPC] (1) Gui Start SreenStream SUCCESS
[qt.network.ssl] QSslSocket: cannot call unresolved function SSL_get_peer_certificate
[UPD] Failed to fetch update information: Network error: SSL handshake failed
```

zhovner, 2022-04-03:
Looks like you have some kind of DPI/[MiTM ](https://en.wikipedia.org/wiki/Man-in-the-middle_attack) system in your network that hijack HTTPS connections, and you get this SSL certificate error. Please try to open https://update.flipperzero.one/builds/firmware/ in your browser. qFlipper downloads firmware from this address.

gsurkov, 2022-04-05:
@NoraCodes You seem to be using the yet unreleased Ubutnu 22.04. As far as I can see, they are [switching to OpenSSL 3.0](https://discourse.ubuntu.com/t/openssl-3-0-transition-plans/24453). qFlipper requires OpenSSL 1.1 which is not (yet) shipped by default with your distro. 

As of now, the `openssl-1.1` package is missing in 22.04, but it's possible to [take it from the current stable release](https://packages.ubuntu.com/impish-updates/libssl1.1) and install it manually, this should fix it.

gsurkov, 2022-04-05:
Short answer: it's on the AppImage exclusion list.
Long(ish) answer: Shipping every possible library within the AppImage would massively increase its size and introduce security issues because of outdated library versions. Certain libraries are considered unconditionally present on all Linux systems, and therefore not necessary to include. 
It works more or less well until a distro decides to break some dependencies. In this particular case, I was completely unaware of this change until you pointed it out. 

I hope that Ubuntu devs will add libssl-1.1 in the final version because many programs still do depend on it, otherwise we will have to include it in the AppImage indeed.

NoraCodes, 2022-04-06:
That entirely makes sense! I will file an issue against the 22.04 beta.

gsurkov, 2022-04-06:
Thanks! It would be nice if they at least added it in the main repository so it would be installable through apt-get.

NoraCodes, 2022-04-06:
Apparently openssl upstream is dropping support for 1.1 next year, so I doubt it will be included in 22.04, as it's an LTS release. Perhaps it would be possible to support 3.0 instead?

probonopd, 2022-04-09:
Sorry for the hassle!

Distributions dropping stuff suddenly at will is becoming an increasing issue. Maybe we should just bundle _all_ libraries in the AppImage (=empty excludelist) and call it a day. Tools like https://github.com/AppImageCrafters/appimage-builder and https://github.com/probonopd/go-appimage/tree/master/src/appimagetool `-s deploy` are going that route.

probonopd, 2022-04-09:
> Thanks! It would be nice if they at least added it in the main repository so it would be installable through apt-get.

Please do open an issue analog to this one:

https://bugs.launchpad.net/ubuntu/+source/ubuntu-meta/+bug/1965636

gsurkov, 2022-04-10:
@NoraCodes It won't be possible to easily transition to 3.0 now because the build is done on Ubuntu 18.04 lts for compatibility reasons, and the Qt packages there rely on openssl 1.1. 
Ubuntu .deb packages are discussed, but we are currenly lacking maintainers for that.

@probonopd As a quick fix, I probably will have to add openssl-1.1 to the AppImage, but bundling everything might not be a good idea depending on the resulting binary size and performance. I have already ran into issues with bundling some libs (e.g. libusb) which led to many weird bugs on different systems. I will experiment with that nevertheless of course and provide a working solution before or shortly after 22.04 release.

gsurkov, 2022-04-27:
Fixed in 1.0.2. Ended up including libssl inside of AppImage bundle.

closed 2022-04-27 by gsurkov