flipperdevices/qFlipper #54: Installation / Update fails on latest release (installer 1.0.01) and installation is corrupted.

cboldon, 2022-04-08:
**Describe the bug**
Opening qFlipper prompted to update to the latest version of qFlipper (installer 1.0.1)

**To Reproduce**
Steps to reproduce the behavior:
1. Download qFlipper 1.0.1 installer
2. Run installer

**Expected behavior**
A successful installation

**Actual behavior**
Installer encounters a problem about 1/3 the way into install. Error message supplied:
"Error opening file for writing"
"[...]\images\arrow-down.png"
[image]

**Additional troubleshooting**
1. If I click "Ignore" on the error dialog, another error message of similar type is displayed referencing another PNG file.

zhovner, 2022-04-09:
Thanks for your report. 
Can you please tell:

1. What version of windows you are running?

2. Do you have admin permissions on your user account?

3 Did you see smart screen warning "this file is running with admin permissions"?

zhovner, 2022-04-09:
Please try this version (choose EXE) https://update.flipperzero.one/builds/qFlipper/nsis_check_admin/

zhovner, 2022-06-06:
Closed due to no response

closed 2022-06-06 by zhovner