flipperdevices/qFlipper #55: Failed to start RPC session update error

anonzak, 2022-04-22:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
I was attempting to perform the first update after receiving the Flipper (to v0.55.1).

**To Reproduce**
Steps to reproduce the behavior:
1. Started qFlipper
2. Plugged in Flipper
3. Hit update button
4. Wait for update to complete

**Expected behavior**
I expected the device firmware to update and databases to be added to the SD card.

**Actual behavior**
qFlipper encounters an error when trying to start the RPC session after the update completes.

**Additional troubleshooting**
1. Error persists after multiple reboots/reconnections/attempts.
2. Error persists after connecting Flipper to computer in Recovery Mode.
3. Error persists while attempting to use any of the firmware releases in qFlipper (RC and Dev)
4. Downloading the firmware file for v0.55.1 from the site and installing from file completes the update successfully, but does not create any databases.

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
5. A screenshot of the application window (if it is a graphical issue).

**Additional context**
When trying to update from qFlipper normally, it seems like the firmware did update (the FW version on my Flipper shows 0.55.1), but the SD card still shows "No Databases". I am using a 128 GB Samsung EVO Select SD card.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
1. Click the "LOGS" button to open the log window.
3. Right click on the log window.
6. In the context menu, select "Browse all logs...".
7. A folder containing several log files should open. 
8. Find the most recent file, this is it!

**Warning!** The log will contain your Flipper's unique name. If you wish not to disclose it, search/replace
all its occurrences before attaching the file.
[qFlipper-20220422-183625.txt](https://github.com/flipperdevices/qFlipper/files/8544810/qFlipper-20220422-183625.txt)

gsurkov, 2022-04-25:
Judging by the log you provided, there is a problem with opening Flipper's serial port. 
Was Flipper connected to the USB port directly or via a hub? 
If it was the latter, could you please connect it directly and try again? In case it was connected directly, try the opposite (connecting it via a USB hub, if available).

anonzak, 2022-04-25:
I attempted to update the firmware using a USB hub and directly connected to the computer, but still received the same error. I was able to get further in the update process (past adding the databases to the SD card) by using a new SD card. The new SD card was the same brand and model as the previous one (I had purchased them at the same time). I still received the same permission denied error at the end of the update.

Just this morning I was able to successfully update my Flipper. I reformatted the SD card and used the qFlipper-cli.exe program to see if that could solve any permissions issues, but my terminal crashed halfway through the update which left my Flipper without any images on the screen and the menus did not seem to work. I disconnected and reconnected my Flipper and opened up the normal qFlipper.exe and used the "repair" option, which seems to have successfully updated the device to the most recent FW version.

Attached are my logs from the repair in case they can help with any troubleshooting for future issues.

[qFlipper-20220425-093139.txt](https://github.com/flipperdevices/qFlipper/files/8555604/qFlipper-20220425-093139.txt)

gsurkov, 2022-04-25:
Thank you for the info!
If the problem persists with subsequent updates, please write another comment here and we will investigate it further.

merchantdanish, 2024-02-27:
i cant update my flipper zero via qflipper and its not connecting to my dell laptop I can only connect in dfu/recovery mode but when I repair it fails at the end. I have attached the most recent log in this. PLEASE HELP ME..
[qFlipper-20240226-200532.txt](https://github.com/flipperdevices/qFlipper/files/14413036/qFlipper-20240226-200532.txt)

merchantdanish, 2024-02-27:
@gsurkov there's not a problem with flipper the problem was with me laptop usb ports i fixed it just needed a driver update thats it now its working fine.