flipperdevices/qFlipper #56: Larger files (11mb) will crash qflipper

Hackerman-the-tired, 2022-04-23:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
What were you trying to do?

I was attempting to push an update package via qflipper (11mb file)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to SD Card in qflipper
2. Right click and select Upload
3. Select file
4. See error

**Expected behavior**
It is expected that the file will upload and process as a package update, but will not allow upload at all.

**Actual behavior**
An error is thrown and the flipper will disconnect

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? - Yes
2. Does reconnecting/rebooting Flipper solve it? - No

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
1. Click the "LOGS" button to open the log window.
2. Right click on the log window.
3. In the context menu, select "Browse all logs...".
4. A folder containing several log files should open. 
6. Find the most recent file, this is it!

**Warning!** The log will contain your Flipper's unique name. If you wish not to disclose it, search/replace
all its occurrences before attaching the file.
[qFlipper-Issue.txt](https://github.com/flipperdevices/qFlipper/files/8546067/qFlipper-Issue.txt)

gsurkov, 2022-04-25:
It is a long-standing issue with large files due to hardware and firmware limitations. Since big files are becoming increasingly common, I think I have an idea how to improve bigger file handling and thus fix the issue. Making it a first priority now.

gsurkov, 2022-04-27:
Should be fixed now in 1.0.2. It is still very slow but at least won't give any timeout errors and will show the progress (for now, only during uploads).

Hackerman-the-tired, 2022-10-11:
Excellent, I'll give it a try soon, thanks for looking into this!!