flipperdevices/qFlipper #57: tags can't be built due to missing files

ZeroChaos-, 2022-04-23:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
Build qFlipper from a github tag tarball based on README.md

**To Reproduce**
Steps to reproduce the behavior:
Download the latest tag (I'm using 1.0.1) and attempt to follow linux build directions

**Expected behavior**
I expect it to not error.

**Actual behavior**
This is wrapped in gentoo's package manager but the error should be fairly stand out
```
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/qflipper-1.0.1/work/qFlipper-1.0.1 ...
 * Running qmake ...
Info: creating stash file /var/tmp/portage/net-wireless/qflipper-1.0.1/work/qFlipper-1.0.1/.qmake.stash                                                               [ ok ]
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-wireless/qflipper-1.0.1/work/qFlipper-1.0.1 ...
make -j1 
cd 3rdparty/ && ( test -e Makefile || /usr/lib64/qt5/bin/qmake -o Makefile /var/tmp/portage/net-wireless/qflipper-1.0.1/work/qFlipper-1.0.1/3rdparty/3rdparty.pro 'QMAKE_AR=x86_64-pc-linux-gnu-ar cqs' QMAKE_CC=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C=x86_64-pc-linux-gnu-gcc QMAKE_LINK_C_SHLIB=x86_64-pc-linux-gnu-gcc QMAKE_CXX=x86_64-pc-linux-gnu-g++ QMAKE_LINK=x86_64-pc-linux-gnu-g++ QMAKE_LINK_SHLIB=x86_64-pc-linux-gnu-g++ QMAKE_OBJCOPY=x86_64-pc-linux-gnu-objcopy QMAKE_RANLIB= QMAKE_STRIP= 'QMAKE_CFLAGS=-Os -march=native -mtune=native -pipe -frecord-gcc-switches' QMAKE_CFLAGS_RELEASE= QMAKE_CFLAGS_DEBUG= 'QMAKE_CXXFLAGS=-Os -march=native -mtune=native -pipe -frecord-gcc-switches' QMAKE_CXXFLAGS_RELEASE= QMAKE_CXXFLAGS_DEBUG= 'QMAKE_LFLAGS=-Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0' QMAKE_LFLAGS_RELEASE= QMAKE_LFLAGS_DEBUG= -spec linux-g++ CONFIG+=qtquickcompiler DEFINES+=DISABLE_APPLICATION_UPDATES ) && make -f Makefile 
fatal: not a git repository (or any parent up to mount point /var/tmp)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /var/tmp)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
fatal: not a git repository (or any parent up to mount point /var/tmp)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
WARNING: Failure to find: nanopb/pb_common.c
WARNING: Failure to find: nanopb/pb_decode.c
WARNING: Failure to find: nanopb/pb_encode.c
make[1]: Entering directory '/var/tmp/portage/net-wireless/qflipper-1.0.1/work/qFlipper-1.0.1/3rdparty'
make[1]: *** No rule to make target 'nanopb/pb_common.c', needed by 'pb_common.o'.  Stop.
make[1]: Leaving directory '/var/tmp/portage/net-wireless/qflipper-1.0.1/work/qFlipper-1.0.1/3rdparty'
make: *** [Makefile:52: sub-3rdparty-make_first] Error 2
```

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?
Yup, it reliably fails to build because the files it's looking for are not in the tag tarball

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
Application can't build, and so it doesn't make logs

**Additional context**
Github tag tarballs don't include submodules, I'm honestly not sure the right fix here but I'll ask around.

ZeroChaos-, 2022-04-23:
As a followup, I can confirm this error doesn't occur if I check out git HEAD.  Which is expected since it fetches the submodules.

gsurkov, 2022-04-26:
These tarballs are automatically generated by Github and indeed don't take submodules into account. It is being worked on and hopefully will be fixed soon.

gsurkov, 2022-08-02:
Turns out, it won't be fixed. It's a built-in Github behaviour and as far as I know, not much can be done about it.
I would gladly learn any (easy) methods for changing it, but until that, a warning in the Known Bugs section is the best I can do.
A good alternative would be checking out the latest tag and building it instead.

closed 2022-08-02 by gsurkov