flipperdevices/qFlipper #59: qFlipper SIGSEGV

ZeroChaos-, 2022-04-24:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
What were you trying to do?
Run qFlipper

**To Reproduce**
Steps to reproduce the behavior:
Build git HEAD on Gentoo following build instructions.

**Expected behavior**
The UI to appear

**Actual behavior**
Crashes with SIGSEGV

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?
Persistent across rebuilds

**Additional context**
```
Starting program: /usr/bin/qFlipper 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
[New Thread 0x7ffff2f56640 (LWP 24891)]
[New Thread 0x7ffff1600640 (LWP 24892)]
[New Thread 0x7ffff0dff640 (LWP 24893)]
[New Thread 0x7fffe3fff640 (LWP 24894)]
[New Thread 0x7fffe37fe640 (LWP 24895)]
[New Thread 0x7fffe2ffd640 (LWP 24897)]
[Thread 0x7fffe2ffd640 (LWP 24897) exited]
[New Thread 0x7fffe2ffd640 (LWP 24898)]
[New Thread 0x7fffe2437640 (LWP 24899)]
[Thread 0x7fffe2ffd640 (LWP 24898) exited]
[New Thread 0x7fffe2ffd640 (LWP 24900)]
[New Thread 0x7fffe1c36640 (LWP 24901)]
[Thread 0x7fffe2437640 (LWP 24899) exited]
[Thread 0x7fffe2ffd640 (LWP 24900) exited]
[Thread 0x7fffe1c36640 (LWP 24901) exited]
[New Thread 0x7fffe1c36640 (LWP 24902)]
[New Thread 0x7fffe2ffd640 (LWP 24904)]
[Thread 0x7fffe1c36640 (LWP 24902) exited]
[Thread 0x7fffe2ffd640 (LWP 24904) exited]
[New Thread 0x7fffe2ffd640 (LWP 24905)]
[New Thread 0x7fffe1c36640 (LWP 24906)]
[New Thread 0x7fffe2437640 (LWP 24907)]
[New Thread 0x7fffe1435640 (LWP 24908)]
[New Thread 0x7fffe0c34640 (LWP 24909)]
[New Thread 0x7fffcbfff640 (LWP 24910)]
[New Thread 0x7fffcb7fe640 (LWP 24911)]
[New Thread 0x7fffca8ab640 (LWP 24912)]
[New Thread 0x7fffca0aa640 (LWP 24913)]
[New Thread 0x7fffc98a9640 (LWP 24914)]
[New Thread 0x7fffc90a8640 (LWP 24915)]

Thread 20 "QQmlThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcb7fe640 (LWP 24911)]
0x00007ffff6fb0ab0 in ?? () from /usr/lib64/libQt5Qml.so.5
(gdb) bt
#0  0x00007ffff6fb0ab0 in  () at /usr/lib64/libQt5Qml.so.5
#1  0x00007ffff6fb0e36 in  () at /usr/lib64/libQt5Qml.so.5
#2  0x00007ffff6faee9f in  () at /usr/lib64/libQt5Qml.so.5
#3  0x00007ffff6faf2df in  () at /usr/lib64/libQt5Qml.so.5
#4  0x00007ffff6faf5e7 in  () at /usr/lib64/libQt5Qml.so.5
#5  0x00007ffff6f42a2f in  () at /usr/lib64/libQt5Qml.so.5
#6  0x00007ffff6f46a86 in  () at /usr/lib64/libQt5Qml.so.5
#7  0x00007ffff6f3bcf5 in QQmlDataBlob::tryDone() () at /usr/lib64/libQt5Qml.so.5
#8  0x00007ffff6f77337 in QQmlTypeLoader::setCachedUnit(QQmlDataBlob*, QV4::CompiledData::Unit const*) () at /usr/lib64/libQt5Qml.so.5
#9  0x00007ffff6f7ad7f in void QQmlTypeLoader::doLoad<CachedLoader>(CachedLoader const&, QQmlDataBlob*, QQmlTypeLoader::Mode) () at /usr/lib64/libQt5Qml.so.5
[... 256 lines cut ...]
#2  0x00007ffff378b307 in xcb_wait_for_event () at /usr/lib64/libxcb.so.1
#3  0x00007ffff31a00cc in  () at /usr/lib64/libQt5XcbQpa.so.5
#4  0x00007ffff68a3b09 in  () at /usr/lib64/libQt5Core.so.5
#5  0x00007ffff638f157 in  () at /lib64/libc.so.6
#6  0x00007ffff6411bbc in  () at /lib64/libc.so.6

Thread 1 (Thread 0x7ffff37457c0 (LWP 24887) "qFlipper"):
#0  0x00007ffff638be9a in  () at /lib64/libc.so.6
#1  0x00007ffff638e540 in pthread_cond_wait () at /lib64/libc.so.6
#2  0x00007ffff68a7fba in  () at /usr/lib64/libQt5Core.so.5
#3  0x00007ffff68a7d09 in QWaitCondition::wait(QMutex*, QDeadlineTimer) () at /usr/lib64/libQt5Core.so.5
#4  0x00007ffff6fc2fd3 in  () at /usr/lib64/libQt5Qml.so.5
#5  0x00007ffff6f7add7 in void QQmlTypeLoader::doLoad<CachedLoader>(CachedLoader const&, QQmlDataBlob*, QQmlTypeLoader::Mode) () at /usr/lib64/libQt5Qml.so.5
#6  0x00007ffff6f77388 in QQmlTypeLoader::loadWithCachedUnit(QQmlDataBlob*, QV4::CompiledData::Unit const*, QQmlTypeLoader::Mode) () at /usr/lib64/libQt5Qml.so.5
#7  0x00007ffff6f7925d in QQmlTypeLoader::getType(QUrl const&, QQmlTypeLoader::Mode) () at /usr/lib64/libQt5Qml.so.5
#8  0x00007ffff6f61caa in QQmlComponentPrivate::loadUrl(QUrl const&, QQmlComponent::CompilationMode) () at /usr/lib64/libQt5Qml.so.5
#9  0x00007ffff6f9d556 in QQmlApplicationEnginePrivate::startLoad(QUrl const&, QByteArray const&, bool) () at /usr/lib64/libQt5Qml.so.5
#10 0x00007ffff6f9d64f in QQmlApplicationEngine::load(QUrl const&) () at /usr/lib64/libQt5Qml.so.5
#11 0x0000555555576e3b in  ()
#12 0x0000555555578d66 in  ()
#13 0x0000555555576206 in  ()
#14 0x00007ffff63302fc in  () at /lib64/libc.so.6
#15 0x00007ffff63303a8 in __libc_start_main () at /lib64/libc.so.6
#16 0x0000555555576301 in  ()
```

gsurkov, 2022-04-25:
Could you please tell what exact version of Qt is installed on your system?
Does the pre-built AppImage version run on it?

ZeroChaos-, 2022-04-25:
dev-qt/qtcore-5.15.3
Yes, the pre-built AppImage does work.

zhovner, 2022-06-06:
@ZeroChaos- do you want to make an eBuild of qFlipper for Gentoo?
Otherwise, we should close this issue and recommend to use AppImage.

ZeroChaos-, 2022-06-06:
That was my goal, yes

NetBUG, 2022-07-07:
Same issue.
```netbug@netbug-mbp2 MacOS % qmake --version
QMake version 3.1
Using Qt version 6.3.1 in /opt/homebrew/lib```

```./qFlipper
zsh: killed     ./qFlipper```

blshkv, 2022-07-14:
hello all, 

the problem is with ```3rdparty``` directory and its nanopb (the only dependency so far).
Basically, that symlink is not getting included in a release tar-ball and that's a good thing. However, the compilation does not really fail when there is no nanopd found.

Currently, qmake scripts expects a bundled nanopb only. I patched it to search in the system libraries instead.
You might want to adopt the patch and make it more universal.