flipperdevices/qFlipper #62: Crash on macOS qFlipper 1.0.1 on M1 Mac Pro (os/threads_posix.h:46)

dweekly, 2022-04-26:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               qFlipper [80002]
Path:                  /Applications/qFlipper.app/Contents/MacOS/qFlipper
Identifier:            test.qFlipper
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-04-25 19:31:45.3184 -0700
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Anonymous UUID:        4EA937FA-C98B-8674-73F1-65BE80031B62

Sleep/Wake UUID:       EFD9ABD2-A45C-48F4-88F0-0FA93A2EA5F1

Time Awake Since Boot: 1800000 seconds
Time Since Wake:       123 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
Assertion failed: (pthread_mutex_lock(mutex) == 0), function usbi_mutex_lock, file ./os/threads_posix.h, line 46.

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x7ff8af48e940 ???
1   libsystem_kernel.dylib        	    0x7ff81ee8300e __pthread_kill + 10
2   libsystem_pthread.dylib       	    0x7ff81eeb91ff pthread_kill + 263
3   libsystem_c.dylib             	    0x7ff81ee04d24 abort + 123
4   libsystem_c.dylib             	    0x7ff81ee040cb __assert_rtn + 314
5   libusb-1.0.0.dylib            	       0x1090a522b 0x109097000 + 57899
6   libusb-1.0.0.dylib            	       0x10909aa40 libusb_ref_device + 48
7   libusb-1.0.0.dylib            	       0x10909b7fe libusb_open + 138
8   qFlipper                      	       0x100815f6f USBDeviceDetector::fillDeviceInfo(USBDeviceInfo const&) + 271
9   qFlipper                      	       0x100816b4b 0x1007a2000 + 478027
10  QtCore                        	       0x10dbdb614 QObject::event(QEvent*) + 900
11  QtWidgets                     	       0x10a6c8a56 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
12  QtWidgets                     	       0x10a6c9e02 QApplication::notify(QObject*, QEvent*) + 466
13  QtCore                        	       0x10dbb22b6 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
14  QtCore                        	       0x10dbb33f3 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 803
15  libqcocoa.dylib               	       0x10d3ee892 0x10d3b2000 + 247954
16  libqcocoa.dylib               	       0x10d3eef98 0x10d3b2000 + 249752
17  CoreFoundation                	    0x7ff81ef7fbca __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
18  CoreFoundation                	    0x7ff81ef7fb32 __CFRunLoopDoSource0 + 180
19  CoreFoundation                	    0x7ff81ef7f8b0 __CFRunLoopDoSources0 + 242
20  CoreFoundation                	    0x7ff81ef7e2c0 __CFRunLoopRun + 892
21  CoreFoundation                	    0x7ff81ef7d884 CFRunLoopRunSpecific + 562
22  HIToolbox                     	    0x7ff828073ce6 RunCurrentEventLoopInMode + 292
23  HIToolbox                     	    0x7ff828073913 ReceiveNextEventCommon + 283
24  HIToolbox                     	    0x7ff8280737e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
25  AppKit                        	    0x7ff8218db53d _DPSNextEvent + 927
26  AppKit                        	    0x7ff8218d9bfa -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
27  AppKit                        	    0x7ff8218cc2a9 -[NSApplication run] + 586
28  libqcocoa.dylib               	       0x10d3edd74 0x10d3b2000 + 245108
29  QtCore                        	       0x10dbae6b7 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
30  QtCore                        	       0x10dbb2852 QCoreApplication::exec() + 130
31  qFlipper                      	       0x1007b0fb3 main + 371
32  dyld                          	       0x2009d451e start + 462

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffdb89a4 0x7ff7ffdb4000 + 18852
[... 471 lines cut ...]
AirPort: Wi-Fi, wl0: Feb  8 2022 01:51:44 version 20.90.39.0.8.7.114 FWID 01-510a2aee
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Thunderbolt Ethernet Slot 0 2, Ethernet, en7
PCI Card: Aquantia AQC107-B0, Ethernet Controller, Thunderbolt@3,0,0
USB Device: USB31Bus
USB Device: USB3.2 Hub
USB Device: USB3.2 Hub
USB Device: USB3.2 Hub
USB Device: USB3.2 Hub
USB Device: USB3.2 Hub
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: DFU in FS Mode
USB Device: USB2.1 Hub
USB Device: Logitech BRIO
USB Device: USB2.1 Hub
USB Device: USB Optical Mouse
USB Device: Yeti X
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Device: Sabrent Ethernet Adapter, Sabrent, 1, 20.1

dweekly, 2022-04-26:
Note: was able to perform a repair after relaunching, was able to flash successfully an update.

Here is the repair log.

[APP] qFlipper version 1.0.1 commit beb47b7f 2022-04-06T06:16:06
[APP] OS info: macOS 10.16 10.16 21.4.0
[REG] Detected new device: VID_0x483:PID_0xdf11
[DBG] Device reported transfer size: 1024
[DBG] Bytes uploaded: 32 100%
[DBG] Upload has finished.
[REG] Registering the device
[BKD] Current device changed to "Shythrn4"
[UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
[UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
[UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
[DEV] Full Repair @Shythrn4 START
[ZIP] Uncompressing file with size of 168696 bytes...
[ZIP] Uncompression finished : No Error
[ZIP] Uncompressing file with size of 26143 bytes...
[ZIP] Uncompression finished : No Error
[RCY] Set Recovery boot mode @Shythrn4 START
[DBG] Device is not idle, resetting...
[DBG] Device reported transfer size: 1024
[DBG] Bytes uploaded: 128 100%
[DBG] Upload has finished.
[DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
[DBG] Device is not idle, resetting...
[DBG] Device reported transfer size: 1024
[DBG] Failed to perform control transfer: LIBUSB_ERROR_NO_DEVICE
[DBG] Unable to get device status
[DBG] An error has occured during download phase
[DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
[DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
[DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
[REG] Device went offline: VID_0x483:PID_0xdf11
[REG] Detected new device: VID_0x483:PID_0xdf11
[DBG] Device reported transfer size: 1024
[DBG] Bytes uploaded: 32 100%
[DBG] Upload has finished.
[REG] Device went back online
[RCY] Set Recovery boot mode @Shythrn4 SUCCESS
[RCY] Co-Processor Firmware Download @Shythrn4 START
[DBG] Device is not idle, resetting...
[DBG] Device reported transfer size: 1024
[DBG] Bytes uploaded: 2 100%
[DBG] Upload has finished.
[DBG] FUS appears NOT to be running: Error occured, Not running.
[DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
[DBG] Device is not idle, resetting...
[DBG] Device reported transfer size: 1024
[DBG] Failed to perform control transfer: LIBUSB_ERROR_NO_DEVICE
[DBG] Bytes uploaded: 0 0%
[DBG] End of transmission.
[DBG] Upload has finished.
[DBG] Failed to read FUS status
[DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
[DBG] vvv ATTENTION: Ignore errors below until matching closing message. vvv
[DBG] ^^^ ATTENTION: Ignore errors above until matching opening message. ^^^
[REG] Device went offline: VID_0x483:PID_0xdf11
[REG] Detected new device: VID_0x483:PID_0xdf11
[DBG] Device reported transfer size: 1024
[DBG] Bytes uploaded: 32 100%
[DBG] Upload has finished.
[REG] Device went back online
[DBG] Device is not idle, resetting...
[REG] Device went offline: VID_0x483:PID_0xdf11
[REG] Detected new device: VID_0x483:PID_0xdf11
[DBG] Device reported transfer size: 1024
[DBG] Bytes uploaded: 32 100%
[DBG] Upload has finished.
[... 2129 lines cut ...]
[RPC] (464) Gui Start SreenStream SUCCESS
[default] Model size of -103 is less than 0
[default] Model size of -12 is less than 0
[DEV] Backup Internal storage @Shythrn4 START
[RPC] (465) Gui Start VirtualDisplay START
[UTL] Backup /int @Shythrn4 START
[RPC] (465) Gui Start VirtualDisplay SUCCESS
[RPC] (466) Storage List @/int START
[RPC] (466) Storage List @/int SUCCESS
[RPC] (467) Storage Read @/int/bt.settings START
[RPC] (467) Storage Read @/int/bt.settings SUCCESS
[RPC] (468) Storage Read @/int/desktop.settings START
[RPC] (468) Storage Read @/int/desktop.settings SUCCESS
[RPC] (469) Storage Read @/int/notification.settings START
[UTL] Backup /int @Shythrn4 SUCCESS
[RPC] (469) Storage Read @/int/notification.settings SUCCESS
[DEV] Backup Internal storage @Shythrn4 SUCCESS
[RPC] (470) Gui ScreenFrame START
[RPC] (470) Gui ScreenFrame SUCCESS
[BKD] Finalized current operation
[SCR] Screen streaming is already running
[RPC] (471) Gui Stop VirtualDisplay START
[RPC] (471) Gui Stop VirtualDisplay SUCCESS
[RPC] (472) Storage Info @/ext START
[RPC] (472) Storage Info @/ext SUCCESS

gsurkov, 2022-04-26:
Could you please specify what were you doing with qFlipper before the crash?
Does the problem persist or it was a one-time occurence?

zhovner, 2022-04-26:
Thank you for the report. Glad you finally flash your device. For now, close until can be reproduced.

Aeron, 2022-08-04:
I got my Flipper delivered a few hours back today. Happily, I turned it on, connected to my MBP (also M1 Max, running macOS 12.5), downloaded qFlipper (1.1.1), clicked “update” (to release 0.63.3), and… got a crash. I tried again and again, three times in a row, with the same outcome. On the fifth time, it made it somehow, when it wasn’t an active window (maybe it’s relevant, maybe not).

Here’s the latest translated report ↓

<details>
<summary>qFlipper Crash Report 2022-08-04-184530 (Translated)</summary>

```text
Process:               qFlipper [8860]
Path:                  /Applications/qFlipper.app/Contents/MacOS/qFlipper
Identifier:            test.qFlipper
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-08-04 18:56:22.8669 +0700
OS Version:            macOS 12.5 (21G72)
Report Version:        12
Anonymous UUID:        2B94FF88-7968-4FDB-8749-906DC78B9FDE

Sleep/Wake UUID:       BBDBAEF2-851C-4F10-BD2E-CBF08EF157D0

Time Awake Since Boot: 780000 seconds
Time Since Wake:       877 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x3837313900000001 -> 0x0000313900000001 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x3837313900000001
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [8860]

VM Region Info: 0x313900000001 is not in any region.  Bytes after previous region: 53639846559746  Bytes before following region: 51432233369599
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   <translation info unavailable>	       0x1026fd3d0 ???
1   libusb-1.0.0.dylib            	       0x10ac135f6 0x10ac11000 + 9718
2   libusb-1.0.0.dylib            	       0x10ac135f6 0x10ac11000 + 9718
3   libusb-1.0.0.dylib            	       0x10ac1403b libusb_open + 63
4   qFlipper                      	       0x1023794ca USBDevice::open() + 58
5   qFlipper                      	       0x10237467e DfuseDevice::beginTransaction() + 14
6   qFlipper                      	       0x102338d5b Flipper::Zero::DFUDeviceInfoHelper::nextStateLogic() + 43
7   QtCore                        	       0x10f757614 QObject::event(QEvent*) + 900
8   QtWidgets                     	       0x10c244a56 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
9   QtWidgets                     	       0x10c245e02 QApplication::notify(QObject*, QEvent*) + 466
10  QtCore                        	       0x10f72e2b6 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
11  QtCore                        	       0x10f72f3f3 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 803
12  libqcocoa.dylib               	       0x10ef6a892 0x10ef2e000 + 247954
13  libqcocoa.dylib               	       0x10ef6af98 0x10ef2e000 + 249752
14  CoreFoundation                	    0x7ff81ad1b37a __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
15  CoreFoundation                	    0x7ff81ad1b2e2 __CFRunLoopDoSource0 + 180
16  CoreFoundation                	    0x7ff81ad1b0c2 __CFRunLoopDoSources0 + 340
17  CoreFoundation                	    0x7ff81ad19a70 __CFRunLoopRun + 892
18  CoreFoundation                	    0x7ff81ad19034 CFRunLoopRunSpecific + 562
19  HIToolbox                     	    0x7ff823e355e6 RunCurrentEventLoopInMode + 292
20  HIToolbox                     	    0x7ff823e35213 ReceiveNextEventCommon + 283
21  HIToolbox                     	    0x7ff823e350e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
22  AppKit                        	    0x7ff81d689fad _DPSNextEvent + 927
23  AppKit                        	    0x7ff81d68866a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
24  AppKit                        	    0x7ff81d67ad19 -[NSApplication run] + 586
[... 222 lines cut ...]
Stack Guard                       56.1M       13 
VM_ALLOCATE                       4880K      147 
VM_ALLOCATE (reserved)              20K        3         reserved VM address space (unallocated)
__CTF                               756        1 
__DATA                            21.7M      391 
__DATA_CONST                      13.7M      201 
__DATA_DIRTY                       602K      112 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       712.3M       60 
__OBJC_RO                         82.8M        1 
__OBJC_RW                         3200K        2 
__TEXT                           279.1M      401 
__UNICODE                          592K        1 
dyld private memory               1024K        1 
mapped file                        5.6G      755 
shared memory                      836K       20 
unshared pmap                     10.5M        7 
===========                     =======  ======= 
TOTAL                              8.4G     2605 
TOTAL, minus reserved VM space     7.2G     2605 
```
</details>

I got all four reports if it’ll help. Let me know if you need those.