flipperdevices/qFlipper #66: qFlipper fail to launch and black screens Windows 10 Version 10.0.19043 Build 19043

leeroywking, 2022-05-02:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!
Thanks for making qFlipper its the first time I've felt confident to make firmware changes and deploy them to a device

**Describe the bug**
What were you trying to do?
Start qFlipper on my windows machine

**To Reproduce**
Steps to reproduce the behavior:
1. Download qFlipper from https://flipperzero.one/update
2. install using either admin or non-admin permissions (both resulted in same issue)
3. Attempt to launch application

**Expected behavior**
I expect the app to launch and be able to flash firmware

**Actual behavior**
 For my system the screens will go black and when display comes back app has not launched

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? - yes and with a fresh install, and with admin permissions on the install and on launch
2. Does reconnecting/rebooting Flipper solve it? - no

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
```
3426 [APP] qFlipper version 1.0.2 commit 9284d92a 2022-04-27T08:52:33
3426 [APP] OS info: Windows 10 Version 2009 10 10.0.19043
3892 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
4057 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
```
I have a bunch of log files that look just like that from attempts to run today the last log file from a successful launch where I used it follows here in case there is something useful from it. 

https://pastebin.com/14jEN1tC - its long sorry

leeroywking, 2022-05-02:
Full system reboot seems to have resolved this I guess? Still no idea what was going on though since it was just qflipper as far as I can tell

zhovner, 2022-05-02:
So reboot resolve the issue?

leeroywking, 2022-05-03:
Yeah rebooting my host OS resolved the issue this time. Just kinda spooky.

closed 2022-05-03 by zhovner

zhovner, 2022-05-04:
Looks like something with your graphics (OpenGL?) subsystem.