flipperdevices/qFlipper #72: 1.0.2 appImage crashes on Debian10

sweharris, 2022-05-16:
```
% ./qFlipper-x86_64-1.0.2.AppImage 
1068 [APP] qFlipper version 1.0.2 commit 9284d92a 2022-04-27T11:52:33
1068 [APP] OS info: Debian GNU/Linux 10 (buster) 10 4.19.0-20-amd64
1758 [USB] Failed to open device
1759 [REG] Detected new device: VID_0x483:PID_0x5740
/tmp/.mount_qFlippGoym92/AppRun.wrapped: line 18:  4363 Segmentation fault      (core dumped) "$this_dir"/usr/bin/qFlipper

% file core
core: ELF 64-bit LSB core file, x86-64, version 1 (SYSV), SVR4-style, from '/tmp/.mount_qFlippGoym92/usr/bin/qFlipper', real uid: 500, effective uid: 500, real gid: 500, effective gid: 500, execfn: '/tmp/.mount_qFlippGoym92/usr/bin/qFlipper', platform: 'x86_64'
```
A new window does flicker open and closed too quickly to see any contents.

(Same image works fine on GalliumOS 3.1, which is a Ubuntu 18.04 respin).

sweharris, 2022-05-26:
But, interestingly, if I build it from source with `docker-compose exec dev ./build_linux.sh` then the resulting AppImage file works.

gsurkov, 2022-06-02:
Very strange, as the docker-built image should be more or less identical to the official one. Ubuntu 18.04 was chosen as the AppImage build environment, so it is compatible at least with that.
Graphical glitches you have described possibly point towards a video driver problem.
I will try to reproduce that, but I don't have high hopes in that regard.

zhovner, 2022-06-06:
Please check the new version 1.1.0 
@sweharris

sweharris, 2022-06-06:
That ran without issues and was able to update my device to the latest dev version without issue.

zhovner, 2022-06-06:
Nice. Thank you.

closed 2022-06-06 by zhovner