flipperdevices/qFlipper #75: Cannot return to main menu after deleting

vuribe1221, 2022-05-20:
Prerequisites: Have a saved Sub-Ghz **Raw** file

Procedures:
Starting from the lock screen
- unlock device
- press down (favorites menu)
- press right (Sub-GHz menu)
- Select a Raw file (press center button) > Run in app
- press Right (More menu) > Press Delete > Delete

File is deleted and are returned to a list of Sub-Ghz files but cannot press back to return to main menu. A reset is required.

gsurkov, 2022-06-02:
Does your report relate to the Flipper firmware or qFlipper (desktop application)? 
If it's the former, then the proper place to report it would be [here](https://github.com/flipperdevices/flipperzero-firmware/issues).

vuribe1221, 2022-06-03:
Yes it was on the physical device. Closing issue.

Thanks

closed 2022-06-03 by vuribe1221