flipperdevices/qFlipper #79: Custom builds are not well handled

SamuelYvon, 2022-05-28:
**Describe the bug**
If you flash a custom firmware version, qFlipper will detect an unknown firmware, but will not suggest an update (it seems to think it's at the latest version)

**To Reproduce**
Steps to reproduce the behavior:
1. Flash a custom build
2. Open qFlipper
3. The update button is grayed out

**Expected behavior**
I expected the update button to be available, an unknown firmware is not equal to the latest

**Actual behavior**
The update button is unavailable

**The report *must* also include at least one of the following:**
A screenshot of the application window:
[image]

**Additional context**

Maybe this is wanted, in that case, feel free to discard this issue. But in my opinion, if the version is detected as unknown, it should suggest and update. I am aware it is possible to force an update through the GUI, but from a usability perspective, I'm not sure this is the best behavior.

zhovner, 2022-05-28:
Thanks, will check this.
Thats because qFlipper cannot check the version. Try to swtich to `dev` channel.

SamuelYvon, 2022-05-28:
> Thanks, will check this.

I would not mind taking care of this if you want (I'd enjoy it)

> Thats because qFlipper cannot check the version. Try to swtich to dev channel.

What I'm gathering is that because the flipper firmware is forked off the dev channel, it's not working with the release version? If I were to make a custom build of release, it would work?

SamuelYvon, 2022-06-02:
I don't have the DFU files with me right now, but you can build latest of https://github.com/SamuelYvon/flipperzero-firmware/tree/gpio-cli

SamuelYvon, 2022-06-03:
[image]

I now have an unknown firmware, but I can update to latest. Seems good to me!