flipperdevices/qFlipper #90: MacOS 12.4 qFlipper SIGSEGV during firmware update

JuliusNmn, 2022-07-18:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
I was updating my flipper. During update, qFlipper crashed. Luckily, my Flipper went into DFU and I could restore without problems. I was running it from the Disk Image directly and didn't have a micro SD inserted.

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               qFlipper [84405]
Path:                  /private/var/folders/*/qFlipper.app/Contents/MacOS/qFlipper
Identifier:            test.qFlipper
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-07-18 10:20:22.8788 +0200
OS Version:            macOS 12.4 (21F79)
Report Version:        12
Anonymous UUID:        FFADA3AB-BC6D-4986-BA5A-B2A832664591

Sleep/Wake UUID:       3569BB89-B589-4376-B83F-069AEE7C7571

Time Awake Since Boot: 1300000 seconds
Time Since Wake:       238 seconds

System Integrity Protection: enabled

Crashed Thread:        13  Thread (pooled)

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00009b8260803630 -> 0x00001b8260803630 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x00009b8260803630
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [84405]

VM Region Info: 0x1b8260803630 is not in any region.  Bytes after previous region: 29765742376497  Bytes before following region: 75306337552848
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0::  Dispatch queue: com.apple.main-thread
0   ???                           	    0x7ff8a739e940 ???
1   libsystem_kernel.dylib        	    0x7ff816de897a mach_msg_trap + 10
2   libsystem_kernel.dylib        	    0x7ff816de8ce8 mach_msg + 56
3   CoreFoundation                	    0x7ff816eeb540 __CFRunLoopServiceMachPort + 319
4   CoreFoundation                	    0x7ff816ee9bd0 __CFRunLoopRun + 1276
5   CoreFoundation                	    0x7ff816ee9014 CFRunLoopRunSpecific + 562
6   HIToolbox                     	    0x7ff81fffd5e6 RunCurrentEventLoopInMode + 292
7   HIToolbox                     	    0x7ff81fffd34a ReceiveNextEventCommon + 594
8   HIToolbox                     	    0x7ff81fffd0e5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
9   AppKit                        	    0x7ff81985f1fd _DPSNextEvent + 927
10  AppKit                        	    0x7ff81985d8ba -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1394
11  AppKit                        	    0x7ff81984ff69 -[NSApplication run] + 586
12  libqcocoa.dylib               	       0x10d658d74 0x10d61d000 + 245108
13  QtCore                        	       0x10de196b7 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
14  QtCore                        	       0x10de1d852 QCoreApplication::exec() + 130
15  qFlipper                      	       0x100a06583 main + 371
16  dyld                          	       0x200d1551e start + 462

Thread 1:: com.apple.rosetta.exceptionserver
0   runtime                       	    0x7ff7ffc2f9c4 0x7ff7ffc2b000 + 18884
1   runtime                       	    0x7ff7ffc3d460 0x7ff7ffc2b000 + 74848
2   runtime                       	    0x7ff7ffc3eed8 0x7ff7ffc2b000 + 81624

[... 403 lines cut ...]
      "deploymentId" : 240000027
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookPro18,3, BootROM 7459.121.3, proc 8:6:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Apr  6 2022 05:55:54 version 20.90.45.0.8.7.118 FWID 01-e7138ff2
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: DFU in FS Mode
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

EdinTC, 2022-07-18:
Hi @JuliusNmn, you need to install a SD card before updating the firmware as stated in the docs: https://docs.flipperzero.one/basics/first-start#MmMZV

Also from that same page:
```
You need the SD card to update
The firmware update will not work if the memory card is not present or not functioning properly.
```

gsurkov, 2022-07-20:
Does it crash every time or was it just this once?

gsurkov, 2022-07-22:
If that happens again, please reopen this issue and attach the corresponding program log.
See [here](https://github.com/flipperdevices/qFlipper/blob/dev/.github/ISSUE_TEMPLATE/bug_report.md#important---getting-the-full-log) for instructions on how to obtain the log.

closed 2022-07-22 by gsurkov