flipperdevices/qFlipper #92: qFlipper crashes when Flipper Zero locks before flashing firmware

stefan131, 2022-07-25:
**Describe the bug**
If you have set a password on your Flipper Zero, and it locks while picking a file to flash with qFlipper, qFlipper will crash after opening the file. It can't see the flipper anymore, so it will probably try to flash a device that is not connected.

**To Reproduce**
Steps to reproduce the behavior:
1. On your Flipper Zero, go to Settings > Desktop 
2. Click on PIN setup and set a pincode (if you haven't already)
3. Set your Auto Lock Time to 10s
4. Open qFlipper on computer
5. Plug in Flipper Zero, unlock it but make sure it's on the screen where it automatically locks
7. Click 'Select file' in qFlipper and look for a firmware file (don't open it yet)
8. Wait for the Flipper Zero to automatically lock, now open the file in qFlipper

**Expected behaviour**
qFlipper asks user to re-connect or unlock Flipper Zero

**Actual behavior**
qFlipper crashes

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? -> yes
2. Does reconnecting/rebooting Flipper solve it? -> You can have an application open to prevent the Flipper Zero from locking

**I'm sorry for not including the log**
qFlipper crashes, so it's not easy to retrieve the logs. I'm also confident that it will not add any value as the problem is very clear, easy to understand, easy to replicate and it's very understandable that it happens.

gsurkov, 2022-07-27:
I can see why this is happening. Definitely not a case we have been testing for. A fix is under way.

gsurkov, 2022-08-02:
Crash fixed in 1.1.1. There is now a check to see whether the device has not been locked or disconnected while the user was choosing a file.
However, there was no special behaviour added (only an error message), because the stock Flipper firmware never auto-locks the device with PIN.

closed 2022-08-02 by gsurkov