flipperdevices/qFlipper #94: Upstream the udev rules

razzeee, 2022-07-27:
Hey as far as I understood the right thing to do with those udev rules would be to upstream them. So they wouldn't be needed here anymore.

The correct way to do that seems to be to add the devices to this https://github.com/systemd/systemd/tree/main/hwdb.d

I'm not sure if it fits in one of those files. Otherwise you would need to add a new one, here's an example PR how to do that. https://github.com/systemd/systemd/pull/22730

kowalski7cc, 2023-01-10:
@razzeee I did open https://github.com/systemd/systemd/pull/24869, but I'll need to switch it to use hwdb, I'll look in the example PR to understand how to do it

razzeee, 2023-01-10:
Thanks for the update, feel free to keep me updated, kinda interested in this, since the people over at platform.io have similar problems (I'm mostly seeing that used in drone stuff)