flipperdevices/qFlipper #95: Debug logging

razzeee, 2022-07-27:
Hey there, 

as I always struggle with AppImage I started a flatpak port yesterday.
You can find the current code here https://github.com/razzeee/one.flipperzero.qFlipper

It starts fine and works, but it's not finding the flipper for some reason. I'm not sure why.
Are there any build options I could use to track that down, like more logging?

gsurkov, 2022-07-28:
All available logs are written into a text file on a per-run basis. The file is available via LOGS->Open Full Log in the application window.
Can the AppImage version detect the Flipper on the same machine?

razzeee, 2022-07-28:
Yes, it can

razzeee, 2022-07-28:
That log seems to be the same, that I get when I start the flatpak from cli. I kinda hoped, that there was a more detailed mode, from skimming the sources.

gsurkov, 2022-07-28:
There aren't any more logs than that, then.
I'm not very familiar with flatpak, but a quick google reveals lots of usb access problems, probably due to some security features.
I will test your package soon and see what I can do on my side, having qFlipper as a flatpak would be pretty nice actually.

razzeee, 2022-07-28:
If you set up your system with the udev rule already it should work. That's my understanding right now. But that's a manual process, it even is for steam. As said the correct approach according to the knowledgeable people in the flatpak matrix channel would be as I suggested in the other issue.

I think it sees that there's something if I have the flipper connected, when I start it. But it seems like it does not recognize it if I connect it after starting the application.
So that might indeed point toward those events?

`LIBUSB_OPTION_LOG_LEVEL` is something I would have change in the code right? it's not an option I can hand to make (sorry for being this clueless)

This is what happens if I have the flipper connected on start of the flatpak
```
> flatpak run one.flipperzero.qFlipper                                                                                                                                                   
QSocketNotifier: Can only be used with threads started with QThread
332 [APP] qFlipper version 1.1.1 commit 9e1b07c9 2022-07-27T17:29:06
332 [APP] OS info: KDE Flatpak runtime 5.15-21.08 5.18.14-1-MANJARO
480 [REG] Detected new device: VID_0x483:PID_0x5740
641 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
4390 [REG] Device initialization failed: Failed to find a suitable serial port
```

I also just realized, that on my manjaro system the flatpak seems to freeze, as soon as I click the `LOGS` button. I think it worked on my fedora without a problem, will double check.

gsurkov, 2022-07-29:
Based on the log you provided, the following can be said:
qFlipper uses USB device serial numbers to keep track of all Flippers connected to the system (there can be more than one)
Once it receives a hotplug event, it gets a list of all serial ports available on the system and tries to match the serial number it got via the event with one of the serial ports from the list.
The "Failed to find a suitable serial port" error means it got the event, but could not detect a serial port with matching USB device serial number.

I have added a bit more debug output around that section, so we could see which ports it checks in the process.
Also there is now a `-u` or `--usb-log-level` command line option which enables libusb log output. Possible values range from 0 (no logging) to 4 (debug level, very spammy). 
The changes are already in the dev branch, you will need libusb >= 1.0.23 for that to work.

razzeee, 2022-07-29:
This is without the `-u`
```
1701 [APP] qFlipper version unknown commit f56294ab 2022-07-29T12:05:17
1702 [APP] OS info: KDE Flatpak runtime 5.15-21.08 5.18.13-200.fc36.x86_64
1717 [REG] Detected new device: VID_0x483:PID_0x5740
1856 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1896 [DBG] Trying serial port  at /dev/ttyACM0
1984 [DBG] Trying serial port  at /dev/ttyACM0
2041 [DBG] Trying serial port  at /dev/ttyACM0
2101 [DBG] Trying serial port  at /dev/ttyACM0
2157 [DBG] Trying serial port  at /dev/ttyACM0
2212 [DBG] Trying serial port  at /dev/ttyACM0
2265 [DBG] Trying serial port  at /dev/ttyACM0
2320 [DBG] Trying serial port  at /dev/ttyACM0
2374 [DBG] Trying serial port  at /dev/ttyACM0
2425 [DBG] Trying serial port  at /dev/ttyACM0
2481 [DBG] Trying serial port  at /dev/ttyACM0
2536 [DBG] Trying serial port  at /dev/ttyACM0
2601 [DBG] Trying serial port  at /dev/ttyACM0
2656 [DBG] Trying serial port  at /dev/ttyACM0
2712 [DBG] Trying serial port  at /dev/ttyACM0
2773 [DBG] Trying serial port  at /dev/ttyACM0
2836 [DBG] Trying serial port  at /dev/ttyACM0
2901 [DBG] Trying serial port  at /dev/ttyACM0
2958 [DBG] Trying serial port  at /dev/ttyACM0
3015 [DBG] Trying serial port  at /dev/ttyACM0
3071 [DBG] Trying serial port  at /dev/ttyACM0
3124 [DBG] Trying serial port  at /dev/ttyACM0
3180 [DBG] Trying serial port  at /dev/ttyACM0
3234 [DBG] Trying serial port  at /dev/ttyACM0
3297 [DBG] Trying serial port  at /dev/ttyACM0
3353 [DBG] Trying serial port  at /dev/ttyACM0
3408 [DBG] Trying serial port  at /dev/ttyACM0
3468 [DBG] Trying serial port  at /dev/ttyACM0
3522 [DBG] Trying serial port  at /dev/ttyACM0
3576 [DBG] Trying serial port  at /dev/ttyACM0
3630 [DBG] Trying serial port  at /dev/ttyACM0
3681 [DBG] Trying serial port  at /dev/ttyACM0
3737 [DBG] Trying serial port  at /dev/ttyACM0
3800 [DBG] Trying serial port  at /dev/ttyACM0
3857 [DBG] Trying serial port  at /dev/ttyACM0
3910 [DBG] Trying serial port  at /dev/ttyACM0
3967 [DBG] Trying serial port  at /dev/ttyACM0
4024 [DBG] Trying serial port  at /dev/ttyACM0
4081 [DBG] Trying serial port  at /dev/ttyACM0
4137 [DBG] Trying serial port  at /dev/ttyACM0
4200 [DBG] Trying serial port  at /dev/ttyACM0
4256 [DBG] Trying serial port  at /dev/ttyACM0
4309 [DBG] Trying serial port  at /dev/ttyACM0
4363 [DBG] Trying serial port  at /dev/ttyACM0
4417 [DBG] Trying serial port  at /dev/ttyACM0
4471 [DBG] Trying serial port  at /dev/ttyACM0
4531 [DBG] Trying serial port  at /dev/ttyACM0
4584 [DBG] Trying serial port  at /dev/ttyACM0
4641 [DBG] Trying serial port  at /dev/ttyACM0
4698 [DBG] Trying serial port  at /dev/ttyACM0
4754 [DBG] Trying serial port  at /dev/ttyACM0
4810 [DBG] Trying serial port  at /dev/ttyACM0
4865 [DBG] Trying serial port  at /dev/ttyACM0
4919 [DBG] Trying serial port  at /dev/ttyACM0
4973 [DBG] Trying serial port  at /dev/ttyACM0
5028 [DBG] Trying serial port  at /dev/ttyACM0
5084 [DBG] Trying serial port  at /dev/ttyACM0
5141 [DBG] Trying serial port  at /dev/ttyACM0
5192 [DBG] Trying serial port  at /dev/ttyACM0
5247 [DBG] Trying serial port  at /dev/ttyACM0
5299 [DBG] Trying serial port  at /dev/ttyACM0
5354 [DBG] Trying serial port  at /dev/ttyACM0
5409 [DBG] Trying serial port  at /dev/ttyACM0
5462 [DBG] Trying serial port  at /dev/ttyACM0
5514 [DBG] Trying serial port  at /dev/ttyACM0
5566 [DBG] Trying serial port  at /dev/ttyACM0
5619 [DBG] Trying serial port  at /dev/ttyACM0
5675 [DBG] Trying serial port  at /dev/ttyACM0
5730 [DBG] Trying serial port  at /dev/ttyACM0
5785 [DBG] Trying serial port  at /dev/ttyACM0
5841 [DBG] Trying serial port  at /dev/ttyACM0
5898 [DBG] Trying serial port  at /dev/ttyACM0
5951 [DBG] Trying serial port  at /dev/ttyACM0
6004 [DBG] Trying serial port  at /dev/ttyACM0
6057 [DBG] Trying serial port  at /dev/ttyACM0
6110 [DBG] Trying serial port  at /dev/ttyACM0
6167 [DBG] Trying serial port  at /dev/ttyACM0
6224 [DBG] Trying serial port  at /dev/ttyACM0
6281 [DBG] Trying serial port  at /dev/ttyACM0
6338 [DBG] Trying serial port  at /dev/ttyACM0
6399 [DBG] Trying serial port  at /dev/ttyACM0
6454 [DBG] Trying serial port  at /dev/ttyACM0
6509 [DBG] Trying serial port  at /dev/ttyACM0
6567 [DBG] Trying serial port  at /dev/ttyACM0
6623 [DBG] Trying serial port  at /dev/ttyACM0
6680 [DBG] Trying serial port  at /dev/ttyACM0
6736 [DBG] Trying serial port  at /dev/ttyACM0
6799 [DBG] Trying serial port  at /dev/ttyACM0
6855 [DBG] Trying serial port  at /dev/ttyACM0
6911 [DBG] Trying serial port  at /dev/ttyACM0
6967 [DBG] Trying serial port  at /dev/ttyACM0
7022 [DBG] Trying serial port  at /dev/ttyACM0
7079 [DBG] Trying serial port  at /dev/ttyACM0
7134 [DBG] Trying serial port  at /dev/ttyACM0
7199 [DBG] Trying serial port  at /dev/ttyACM0
7259 [DBG] Trying serial port  at /dev/ttyACM0
7315 [DBG] Trying serial port  at /dev/ttyACM0
7371 [DBG] Trying serial port  at /dev/ttyACM0
7426 [DBG] Trying serial port  at /dev/ttyACM0
7443 [REG] Device initialization failed: Failed to find a suitable serial port

```

```
ls -l ttyACM0
crw-rw-rw-+ 1 nfsnobody nfsnobody 166, 0 Jul 29 12:36 ttyACM0
```

So it seems like it's something on the flatpak side I need to figure out.

gsurkov, 2022-07-29:
Looks like the serial port does not receive a USB serial number. qFlipper performs multiple attempts (as the serial number can be loaded by the OS asynchronously), but gives up eventually.

gsurkov, 2022-08-02:
Since the additional debug output was added and the cause of the problem was pinpointed to be outside of qFlipper, closing the issue. Please feel free to ask further questions/report bugs.

closed 2022-08-02 by gsurkov

razzeee, 2022-08-03:
For reference

```
╰─ flatpak run one.flipperzero.qFlipper --usb-log-level 4                                                                                                           ─╯
QSocketNotifier: Can only be used with threads started with QThread
Gtk-Message: 00:40:57.596: Failed to load module "canberra-gtk-module"
Gtk-Message: 00:40:57.597: Failed to load module "canberra-gtk-module"
624 [APP] qFlipper version unknown commit cced2243 2022-08-02T20:52:35
624 [APP] OS info: KDE Flatpak runtime 5.15-21.08 5.18.14-1-MANJARO
628 [USB] [DBG] [timestamp] [threadID] facility level [function call] <message>
628 [USB] [DBG] --------------------------------------------------------------------------------
628 [USB] [DBG] [ 0.706114] [00000002] libusb: debug [libusb_get_device_descriptor]  
628 [USB] [DBG] [ 0.706142] [00000002] libusb: debug [libusb_open] open 1.2
741 [USB] [DBG] [ 0.819461] [00000002] libusb: debug [usbi_add_event_source] add fd 15 events 4
741 [USB] [DBG] [ 0.819551] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x56017e05dc30
741 [USB] [DBG] [ 0.819584] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
741 [USB] [DBG] [ 0.819617] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
741 [USB] [DBG] [ 0.819641] [00000002] libusb: debug [handle_events] event sources modified, reallocating event data
741 [USB] [DBG] [ 0.819665] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
744 [USB] [DBG] [ 0.822449] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
744 [USB] [DBG] [ 0.822511] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
744 [USB] [DBG] [ 0.822538] [00000002] libusb: debug [handle_control_completion] handling completion status 0
744 [USB] [DBG] [ 0.822569] [00000002] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
744 [USB] [DBG] [ 0.822594] [00000002] libusb: debug [usbi_handle_transfer_completion] transfer 0x56017e05dc30 has callback 0x7fb7e55b0650
744 [USB] [DBG] [ 0.822620] [00000002] libusb: debug [sync_transfer_cb] actual_length=4
744 [USB] [DBG] [ 0.822647] [00000002] libusb: debug [libusb_free_transfer] transfer 0x56017e05dc30
744 [USB] [DBG] [ 0.822673] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x56017e05df80
744 [USB] [DBG] [ 0.822697] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
744 [USB] [DBG] [ 0.822732] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
744 [USB] [DBG] [ 0.822758] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
752 [USB] [DBG] [ 0.830461] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
752 [USB] [DBG] [ 0.830575] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=42
752 [USB] [DBG] [ 0.830603] [00000002] libusb: debug [handle_control_completion] handling completion status 0
752 [USB] [DBG] [ 0.830627] [00000002] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
752 [USB] [DBG] [ 0.830653] [00000002] libusb: debug [usbi_handle_transfer_completion] transfer 0x56017e05df80 has callback 0x7fb7e55b0650
752 [USB] [DBG] [ 0.830682] [00000002] libusb: debug [sync_transfer_cb] actual_length=42
752 [USB] [DBG] [ 0.830710] [00000002] libusb: debug [libusb_free_transfer] transfer 0x56017e05df80
752 [USB] [DBG] [ 0.830745] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x56017e05dc30
752 [USB] [DBG] [ 0.830770] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
752 [USB] [DBG] [ 0.830809] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
752 [USB] [DBG] [ 0.830836] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
755 [USB] [DBG] [ 0.833461] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
755 [USB] [DBG] [ 0.833549] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=4
755 [USB] [DBG] [ 0.833575] [00000002] libusb: debug [handle_control_completion] handling completion status 0
755 [USB] [DBG] [ 0.833599] [00000002] libusb: debug [arm_timer_for_next_timeout] no timeouts, disarming timer
755 [USB] [DBG] [ 0.833626] [00000002] libusb: debug [usbi_handle_transfer_completion] transfer 0x56017e05dc30 has callback 0x7fb7e55b0650
755 [USB] [DBG] [ 0.833670] [00000002] libusb: debug [sync_transfer_cb] actual_length=4
755 [USB] [DBG] [ 0.833697] [00000002] libusb: debug [libusb_free_transfer] transfer 0x56017e05dc30
755 [USB] [DBG] [ 0.833723] [00000002] libusb: debug [libusb_submit_transfer] transfer 0x56017e05df80
755 [USB] [DBG] [ 0.833747] [00000002] libusb: debug [add_to_flying_list] arm timer for timeout in 1000ms (first in line)
755 [USB] [DBG] [ 0.833784] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
755 [USB] [DBG] [ 0.833809] [00000002] libusb: debug [usbi_wait_for_events] poll() 3 fds with timeout in 60000ms
762 [USB] [DBG] [ 0.840454] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 1
762 [USB] [DBG] [ 0.840547] [00000002] libusb: debug [reap_for_handle] urb type=2 status=0 transferred=32
762 [USB] [DBG] [ 0.840572] [00000002] libusb: debug [handle_control_completion] handling completion status 0
[... cut ...]
7572 [USB] [DBG] [ 7.650151] [00000002] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
7572 [USB] [DBG] [ 7.650255] [00000002] libusb: debug [usbi_wait_for_events] poll() 2 fds with timeout in 10ms
7582 [USB] [DBG] [ 7.660345] [00000002] libusb: debug [usbi_wait_for_events] poll() returned 0
```

So as we could see from the previous log, it's probably `QSerialPortInfo::availablePorts();` failing and somehow returning a bunch of objects without `serialnumber`

razzeee, 2022-08-03:
Okay, got it to work now running the flatpak via

```
flatpak run --filesystem=/run/udev:ro one.flipperzero.qFlipper
```

As far as I could tell, the interface works and I was able to flash the firmware successfully.

However the process was not perfect, as the flipper wasn't re-detected on reboot. Which is probably the same issue I described above, when flipper does not show up, if you don't have it connected on app start.

razzeee, 2022-08-03:
seems to have been discussed in 2019 here https://lists.freedesktop.org/archives/flatpak/2019-December/001863.html and I guess that didn't change

I guess patching libusb to handle it like alex suggested, to get hotplug events there might not really be a solution :/