flipperdevices/qFlipper #97: qFlipper on arm64 (raspi4)

akervern, 2022-08-02:
Hey guys, I'm trying to get qFlipper up and running on arm64 but I think I'll need some help.

I did try to stick to the current build chain and adapt it, for now I target to just compile and run, without packaging eerything as an `appimage`.

I commented `linuxdeploy` install and adapted apt repository to have a similar arm64 apt repository.
so I changed a little bit the `dockerfile`, as follow:
```diff
--- a/docker/Dockerfile
+++ b/docker/Dockerfile
@@ -6,7 +6,7 @@ RUN apt-get update && \
     DEBIAN_FRONTEND=noninteractive apt-get install -y \
     software-properties-common
 
-RUN add-apt-repository ppa:beineri/opt-qt-5.15.2-bionic && \
+RUN add-apt-repository ppa:theofficialgman/opt-qt-5.15.2-bionic-arm && \
     apt-get update && \
     DEBIAN_FRONTEND=noninteractive apt-get install -y \
     wget \
@@ -31,11 +31,11 @@ RUN add-apt-repository ppa:beineri/opt-qt-5.15.2-bionic && \
     qt515wayland \
     && apt-get clean && rm -rf /var/lib/apt/lists/* /tmp/* /var/tmp/*
 
-RUN wget https://github.com/linuxdeploy/linuxdeploy/releases/download/continuous/linuxdeploy-x86_64.AppImage -O /usr/bin/linuxdeploy \
-    && chmod +x /usr/bin/linuxdeploy
+# RUN wget https://github.com/linuxdeploy/linuxdeploy/releases/download/continuous/linuxdeploy-x86_64.AppImage -O /usr/bin/linuxdeploy \
+#     && chmod +x /usr/bin/linuxdeploy
 
-RUN wget https://github.com/linuxdeploy/linuxdeploy-plugin-qt/releases/download/continuous/linuxdeploy-plugin-qt-x86_64.AppImage -O /usr/bin/linuxdeploy-plugin-qt \
-    && chmod +x /usr/bin/linuxdeploy-plugin-qt
+# RUN wget https://github.com/linuxdeploy/linuxdeploy-plugin-qt/releases/download/continuous/linuxdeploy-plugin-qt-x86_64.AppImage -O /usr/bin/linuxdeploy-plugin-qt \
+#     && chmod +x /usr/bin/linuxdeploy-plugin-qt
 
 RUN git config --global --replace-all safe.directory '*'
```

Then a created a dedicated arm64 build file:
```bash
#!/bin/bash

set -ex

TARGET='qFlipper'
BUILDDIR='build'
APPDIR_PREFIX="$PWD/$BUILDDIR/AppDir/usr"

export QML_SOURCES_PATHS='..'

mkdir -p $BUILDDIR && cd $BUILDDIR
qmake ../$TARGET.pro -spec linux-g++ CONFIG+=qtquickcompiler PREFIX=$APPDIR_PREFIX && make qmake_all && make -j$(nproc) && make install
```
(dropping few lines tied to `linuxdeploy` execution)

Container is built just fine, build script completed; then I tried to run the build from the raspi4, and after installing a few extra dependencies (not sure about the exact list, but I'll share it later when everything will work) I get the qFlipper window working!

Thus, my issue is now nor qflipper, nor qflipper-cli are able to connect to my device. Only available logs I was able to get are
```
[APP] qFlipper version 1.1.1 commit 9e1b07c9 2022-07-27T17:29:06
[APP] OS info: Debian GNU/Linux 11 (bullseye) 11 5.15.32-v8+
[RPC] Starting RPC session...
[RPC] Failed to start RPC session: Failed to begin CLI session: Device is not responding
```

And now Im stuck how I can fix this connection issue. Do you have any idea what I would need to try/install missing lib to move forward?

Thanks or your help!

gsurkov, 2022-08-02:
Have you set up the [device access rules](https://github.com/flipperdevices/qFlipper/blob/dev/installer-assets/udev/42-flipperzero.rules)?

akervern, 2022-08-05:
Good catch, I missed that part - Thank for your help, now everything seems o work just fine.

closed 2022-08-05 by akervern