flipperdevices/qFlipper #100: Crashes on M1 Mac at the end of firmware flashing

Aeron, 2022-08-04:
I got my Flipper delivered a few hours back today. Happily, I turned it on, connected to my MBP (M1 Max, running macOS 12.5), downloaded and launched qFlipper (1.1.1), and clicked “update” (to release 0.63.3), and… got a crash on 100%. I tried to repair it, again and again, three times in a row, with the same outcome. On the fifth time, it made it somehow, when it wasn’t an active window (maybe it’s relevant, maybe not). Also, I didn’t try rebooting or reconnecting my Flipper because it was in recovery mode anyway.

<details><summary>More structured issue answers, closer to the template</summary>

**Steps to reproduce the behavior**
1. Get a new Flipper
2. Turn it on
3. Connect it via USB to a M1-powered macOS-12.5-running machine
4. Launch qFlipper 1.1.1
5. Change nothing
6. Click “Update”
7. Wait until 100% and after

**Expected behavior**
A finished firmware update process, a running application, and a happy dolphin.

**Actual behavior**
A crashed application and a Flipper device in recovery mode.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? **No**
2. Does reconnecting/rebooting Flipper solve it? **Unknown**

</details>

Here’s the latest crash report. I got all four reports if it’ll help. Let me know if you need those as well.

<details><summary>Crash Report 2022-08-04-185623 (Translated)</summary>

```text
Process:               qFlipper [8860]
Path:                  /Applications/qFlipper.app/Contents/MacOS/qFlipper
Identifier:            test.qFlipper
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-08-04 18:56:22.8669 +0700
OS Version:            macOS 12.5 (21G72)
Report Version:        12
Anonymous UUID:        2B94FF88-7968-4FDB-8749-906DC78B9FDE

Sleep/Wake UUID:       BBDBAEF2-851C-4F10-BD2E-CBF08EF157D0

Time Awake Since Boot: 780000 seconds
Time Since Wake:       877 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x3837313900000001 -> 0x0000313900000001 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0x3837313900000001
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [8860]

VM Region Info: 0x313900000001 is not in any region.  Bytes after previous region: 53639846559746  Bytes before following region: 51432233369599
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)
--->  GAP OF 0x5f9000000000 BYTES
      MALLOC_NANO              600000000000-600008000000 [128.0M] rw-/rwx SM=PRV  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   <translation info unavailable>         0x1026fd3d0 ???
1   libusb-1.0.0.dylib                     0x10ac135f6 0x10ac11000 + 9718
2   libusb-1.0.0.dylib                     0x10ac135f6 0x10ac11000 + 9718
[... 1119 lines cut ...]
377259 [RPC] (6) Gui Stop VirtualDisplay SUCCESS
377259 [RPC] (7) Gui Start SreenStream START
377318 [RPC] (7) Gui Start SreenStream SUCCESS
377318 [RPC] (8) Storage Info @/ext START
377383 [RPC] (8) Storage Info @/ext SUCCESS
580485 [RPC] (9) Storage Info @/ext START
580552 [RPC] (9) Storage Info @/ext SUCCESS
580559 [RPC] (10) Gui Stop ScreenStream START
580622 [RPC] (10) Gui Stop ScreenStream SUCCESS
583067 [RPC] (11) Storage List @/int START
583127 [RPC] (11) Storage List @/int SUCCESS
584401 [RPC] (12) Gui Start SreenStream START
584453 [RPC] (12) Gui Start SreenStream SUCCESS
600641 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
900794 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1200599 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1500666 [default] QIODevice::write (QTcpSocket): device not open
1501449 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1800670 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
2100811 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
2370561 [APP] qFlipper exited
2370600 [default] "There are still \"1\" items in the process of being created at engine destruction."
```

</details>

gsurkov, 2022-08-04:
Thank you for providing all of the crash logs. There is a pattern in them worth investigating. There is a chance that it is somehow connected to Rosetta though, an can truly be fixed only by migrating qFlipper to Qt6 series (which is planned, but cannot be done very easily and will impact users with older hardware/OSes).
The good thing is now that the firmware is updated, it supports the newer update method, which is much more stable, so this should not happen anymore.

montali, 2022-08-10:
Same thing is happening on mine. Tried on my older i7 Macbook, still crashing.

tnj, 2022-08-11:
The same crash happened to me. I tried more than five times but had no luck. Changing the update channel to dev (`a39111ad` when I tried) didn't help for the first time but produced a different crash report, so I tried again then it worked. 

<details><summary>Crash Report</summary>

```
Process:               qFlipper [89370]
Path:                  /Applications/qFlipper.app/Contents/MacOS/qFlipper
Identifier:            test.qFlipper
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-08-11 22:55:30.6101 +0900
OS Version:            macOS 12.1 (21C52)
Report Version:        12
Anonymous UUID:        213E1498-4FBB-6628-4E4B-2192013A45C3

Sleep/Wake UUID:       BE93BB51-CDE7-4235-B672-74E26C306B52

Time Awake Since Boot: 3500000 seconds
Time Since Wake:       564854 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0xe000000000000000 -> 0x0000000000000000 (possible pointer authentication failure)
Exception Codes:       0x0000000000000001, 0xe000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [89370]

VM Region Info: 0 is not in any region.  Bytes before following region: 4306153472
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100aab000-100bd7000    [ 1200K] r-x/r-x SM=COW  ...acOS/qFlipper

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libusb-1.0.0.dylib            	       0x1093c237d 0x1093c0000 + 9085
1   libusb-1.0.0.dylib            	       0x1093c4830 libusb_get_device_descriptor + 43
2   qFlipper                      	       0x100b2651f USBDeviceDetector::fillDeviceInfo(USBDeviceInfo const&) + 207
3   qFlipper                      	       0x100b2714b 0x100aab000 + 508235
4   QtCore                        	       0x10df06614 QObject::event(QEvent*) + 900
5   QtWidgets                     	       0x10a9f3a56 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 262
6   QtWidgets                     	       0x10a9f4e02 QApplication::notify(QObject*, QEvent*) + 466
7   QtCore                        	       0x10dedd2b6 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 166
8   QtCore                        	       0x10dede3f3 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 803
9   libqcocoa.dylib               	       0x10d719892 0x10d6dd000 + 247954
10  libqcocoa.dylib               	       0x10d719f98 0x10d6dd000 + 249752
11  CoreFoundation                	    0x7ff80de03c08 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
12  CoreFoundation                	    0x7ff80de03b70 __CFRunLoopDoSource0 + 180
13  CoreFoundation                	    0x7ff80de038e3 __CFRunLoopDoSources0 + 242
14  CoreFoundation                	    0x7ff80de022ff __CFRunLoopRun + 897
15  CoreFoundation                	    0x7ff80de018a9 CFRunLoopRunSpecific + 567
16  HIToolbox                     	    0x7ff816e834f1 RunCurrentEventLoopInMode + 292
17  HIToolbox                     	    0x7ff816e83247 ReceiveNextEventCommon + 587
18  HIToolbox                     	    0x7ff816e82fe5 _BlockUntilNextEventMatchingListInModeWithFilter + 70
19  AppKit                        	    0x7ff810757b4c _DPSNextEvent + 886
20  AppKit                        	    0x7ff8107561b8 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1411
21  AppKit                        	    0x7ff8107485a9 -[NSApplication run] + 586
22  libqcocoa.dylib               	       0x10d718d74 0x10d6dd000 + 245108
23  QtCore                        	       0x10ded96b7 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 471
24  QtCore                        	       0x10dedd852 QCoreApplication::exec() + 130
25  qFlipper                      	       0x100aba8c3 main + 371
26  dyld                          	       0x200ef64fe start + 462

[... 199 lines cut ...]
STACK GUARD                         12K        3 
Stack                             23.1M       16 
Stack Guard                       56.1M       13 
VM_ALLOCATE                       6752K      160 
VM_ALLOCATE (reserved)              20K        3         reserved VM address space (unallocated)
__DATA                            21.6M      385 
__DATA_CONST                      13.6M      201 
__DATA_DIRTY                       606K      113 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       708.4M       61 
__OBJC_RO                         81.8M        1 
__OBJC_RW                         3136K        3 
__TEXT                           277.2M      397 
__UNICODE                          588K        1 
dyld private memory               1024K        1 
mapped file                        5.5G      740 
shared memory                     1060K       16 
unshared pmap                     8608K        5 
===========                     =======  ======= 
TOTAL                              7.1G     2580 
TOTAL, minus reserved VM space     7.0G     2580 
```

</details>

bertjwregeer, 2022-08-12:
I was able to use qFlipper version 1.0.2 to do the initial update flash of my Flipper, after that I was able to use the latest version of qFlipper without issues to upload newer firmware or alternate firmwares on my flipper. Might be an option to downgrade qFlipper first (ignore the "this is not the latest version" warning).

rubyroobs, 2022-08-12:
Thank you @bertjwregeer !!! Using 1.0.2 ([download link](https://update.flipperzero.one/builds/qFlipper/1.0.2/)) I was also able to get mine flashed and then just tested again in latest qFlipper that I could install the RC ok for 0.64.0. Hopefully others with the same issue see this.

zhovner, 2022-08-12:
Working on this issue, will release separate mac M1 version of qFlipper.

jk779, 2022-08-17:
I just got my Flipper today and the same issue happend to me too, **but** I'm using a Intel Mac?! (Macmini8,1) 
The workaround with Version 1.0.2 worked for me as well :) Maybe it's a problem with the MacOS version as a whole? 🤔

gsurkov, 2022-08-18:
Since the majority of the crashes seem to happen inside libusb, and also some evidence that downgrading the application version is helping, there is a chance that it was at least partially caused by upgrading to libusb 1.0.26 on the build server which happened mid June.

Just to be safe, I have rolled it back to 1.0.24 and the upcoming qFlipper version will be bundled with libusb 1.0.24.
Of course, this is not a proper fix, Rosetta and possible libusb-related problems (bugs and/or improper use) still aren't off the table.

Hope this helps. Everything said earlier still stands: Qt6 and the native M1 version are a priority, which indirectly will help fixing problems like this one.

sudocurse, 2023-11-10:
Not sure if this is just a quirk of my update failure case though (stuck at "Updating resources" even after reboot) but I tried repairing with 1.3.2 and also got this crash, however when i tried 1.0.2:
```
6892 [ZIP] Uncompressing file with size of 168264 bytes...
6894 [ZIP] Uncompression finished : No Error
7311 [ZIP] Uncompressing file with size of 344395 bytes...
7325 [ZIP] Uncompression finished : No Error
7327 [DBG] FileInfo not found
7328 [DEV] Full Repair @Y0nspri0 ERROR: Failed to fetch the files: Failed to fetch file: Failed to open file for writing: Is a directory.
7328 [BKD] Current operation finished with error: "Failed to fetch the files: Failed to fetch file: Failed to open file for writing: Is a directory."
```

closed 2026-09-28 by Aeron