flipperdevices/qFlipper #103: Allow to read Release Notes while performing flash

Altair31, 2022-08-15:
It would be great if we are able to read the release note while upgrading our flipper.
For people like me who will update no matter what is in the patch, it would be interesting to use the update duration to read it.

zhovner, 2022-08-15:
You can read it while updating via mobile app https://flpr.app/
[image]

closed 2022-08-15 by zhovner

Altair31, 2022-08-15:
Why are you talking about mobile app when I'm posting on desktop app.
Mobile app is a real pain to provision firmware to the device (very slow due to BT speed rate) so I still use qFlipper desktop application.
Does the desktop app have the same hidden functionality?

zhovner, 2022-08-15:
>Does the desktop app have the same hidden functionality?

Yeah, you can press on release version and get the changelog
[image]

>Why are you talking about mobile app when I'm posting on desktop app tracker?

Because soon you will be enforced to use mobile app for additional features like NFC attacks and so on, so get used to it :)