flipperdevices/qFlipper #105: Update fails

incyi, 2022-08-18:
**Describe the bug**
I open the qFlipper app on my laptop, it gives an error after attempting an update.

**To Reproduce**
Steps to reproduce the behavior:
1. Open qFlipper app.
2. Wait for update pop-up.
3. Press on update.
4. See going "downloading update till 100%"
5. Wait
6. See error "Update failed"

**Expected behavior**
The expectation is that it will update to the latest qflipper app.

**Actual behavior**
Update fails after some time.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?
- yes
2. Does reconnecting/rebooting Flipper solve it?
- no, it also does not work when my Flipper is not connected.

**Full log:**
4004 [APP] qFlipper version 1.1.2 commit 78c52175 2022-08-09T15:10:36
4005 [APP] OS info: Windows 10 Version 2009 10 10.0.19044
4089 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
4091 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
5535 [SFU] Downloading the application update package...
17405 [SFU] Application update package has been downloaded.
17459 [default] ShellExecute 'c:/Users/***/AppData/Local/Temp/qFlipperSetup-64bit-1.1.3.exe' failed (error 2).
17459 [SFU] Failed to start the application update process.
21561 [APP] qFlipper exited
21894 [default] "There are still \"1\" items in the process of being created at engine destruction."

incyi, 2022-08-18:
This error comes from here, I think. Something wrong with the path ?
https://github.com/flipperdevices/qFlipper/blob/dev/application/applicationupdater.cpp

I have a space in my user name. Can that be an issue ?

gsurkov, 2022-08-18:
Is this the first qFlipper update on your system or were the previous times successful?

incyi, 2022-08-18:
It is the second version that I tried this. 

The previous one also had the same issue. (But I don't remember the version number from thet one) I uninstalled it and then installed this version. But now it happens again.

gsurkov, 2022-08-18:
Are you able to [install the new version manually](https://flipperzero.one/update)? I.e. does the manual installation work on your system?

Maybe the space in the user name has something to do with your problems, we will test it soon.

incyi, 2022-08-18:
Manual install works. Just tried, it is now this version:

[APP] qFlipper version 1.1.3 commit f9bd612e 2022-08-17T16:23:02
[APP] OS info: Windows 10 Version 2009 10 10.0.19044

gsurkov, 2022-08-19:
The space indeed was the culprit. The fix is in the dev branch now, but unfortunately you'll have to install the next release manually for obvious reasons. From then on, it should work smoothly.

closed 2022-08-19 by gsurkov