flipperdevices/qFlipper #109: Minimize button does not work on macOS Monterey

d1str0, 2022-08-26:
Yellow minimize button on UI does not trigger minimization. 

**To Reproduce**
Update to latest qFlipper (1.1.3), run app, try to minimize it

**Expected behavior**
I expect it to fold down to the tray.

**Actual behavior**
Does not respond at all

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run.

```748 [APP] qFlipper version 1.1.3 commit f9bd612e 2022-08-17T07:23:02
748 [APP] OS info: macOS 10.16 10.16 21.5.0
778 [REG] Detected new device: VID_0x483:PID_0x5740
986 [DBG] Trying serial port  at /dev/cu.wlan-debug
986 [DBG] Trying serial port  at /dev/tty.wlan-debug
986 [DBG] Trying serial port  at /dev/cu.Bluetooth-Incoming-Port
986 [DBG] Trying serial port  at /dev/tty.Bluetooth-Incoming-Port
986 [DBG] Trying serial port flip_Hadid33 at /dev/cu.usbmodemflip_Hadid331
986 [DBG] Using  serial port flip_Hadid33 at /dev/cu.usbmodemflip_Hadid331
986 [RPC] Starting RPC session...
1040 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1054 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1130 [RPC] RPC session started successfully.
1131 [RPC] (1) System Device Info START
1240 [RPC] (1) System Device Info SUCCESS
1240 [RPC] (2) Storage Info @/ext START
1293 [RPC] (2) Storage Info @/ext SUCCESS
1293 [RPC] (3) Storage Stat @/ext/Manifest START
1346 [RPC] (3) Storage Stat @/ext/Manifest SUCCESS
1346 [RPC] (4) System Get DateTime START
1398 [DBG] Flipper time skew is -515 milliseconds
1398 [RPC] (4) System Get DateTime SUCCESS
1399 [RPC] (5) System Set DateTime START
1460 [RPC] (5) System Set DateTime SUCCESS
1461 [RPC] Stopping RPC session...
1472 [RPC] Attempting to unload protobuf plugin...
1476 [RPC] Unloaded protobuf plugin.
1476 [RPC] RPC session stopped successfully.
1476 [REG] Registering the device
1477 [DEV] Version: 0.65.3 commit: acc6047e4 radio: 1.13.3
1477 [BKD] Current device changed to "Seamus"
1711 [RPC] Starting RPC session...
1835 [RPC] RPC session started successfully.
1835 [RPC] (1) Gui Start ScreenStream START
1891 [RPC] (1) Gui Start ScreenStream SUCCESS
```

2. A screenshot of the application window (if it is a graphical issue).
[image]

gsurkov, 2022-08-29:
Known bug, for some reason the `showMinimized()` function doesn't  work on MacOS. Might find a workaround sometime later.

gsurkov, 2022-09-21:
Should be fixed in 1.2.0.

closed 2022-09-21 by gsurkov