flipperdevices/qFlipper #112: File browser: can not enter and exit folders when browsing

agathakazar, 2022-08-29:
**Describe the bug**
When using the file browser I can use the arrow keys to switch between different files in the 4 directions. But I can not enter a folder when pressing 'Enter' or 'z' and I can not move up one level when pressing 'Backspace' or 'x'

**To Reproduce**
Steps to reproduce the behavior:
1. Go to File manager
2. Try navigating the menu with the keyboard
3. Try pressing the above mentioned buttons 
4. See that you can not navigate inside and outside of the folders

**Expected behavior**
I expected to be able to navigate the same way I navigate Flipper menus when screen streaming. 

**Actual behavior**
I can't navigate because the buttons  'Enter', 'z',  'Backspace' and 'x' don't work. 

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?
- yes
2. Does reconnecting/rebooting Flipper solve it?
- no

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).
[image]

**Additional context**
The < and > buttons work wen navigating, but pressing them with the mouse is inconvenient.

gsurkov, 2022-08-30:
Frankly speaking, the fact that it's sometimes possible to use keyboard navigation in the file manager is not an intended behaviour, but rather just a property of the GridView Qml component used in its implementation. 

Although keyboard navigation sounds like a logical thing, from my experience it's quite tricky to make that work in Qml. The components tend to lose focus for whatever reason and the behaviour varies from platform to platform. (It took some hacks to even get the D-Pad from screen streaming working.)

This said, I can't promise that it will be implemented, but it is worth trying sometime in the future.

Smooklu, 2022-12-15:
This should be working now! Let us know if you experience any issues :)

closed 2023-01-23 by gsurkov