flipperdevices/qFlipper #120: Building for 32-bit throws a struct packing assertion

RunningDroid, 2022-09-17:
**Describe the bug**
Building with the `-m32` argument in CFLAGS & CXXFLAGS causes the build to throw a struct packing assertion

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this repo & init the submodules
2. `mkdir build && cd build`
3. `qmake ../qFlipper.pro 'QMAKE_CFLAGS=-m32' 'QMAKE_CXXFLAGS=-m32'`
4. `make qmake_all && make`

**Expected behavior**
The build should either fail and say 32-bit builds are unsupported or compile successfully

**Actual behavior**
The build fails with this error:
```
In file included from ../../dfu/device/stm32wb55/stm32wb55.cpp:7:
../../dfu/device/stm32wb55/infotable.h:78:43: error: static assertion failed: Check your struct packing!
   78 | static_assert (sizeof(FUSDeviceInfoTable) == 56, "Check your struct packing!");
      |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
```

**Additional context**
I came across this while updating the Void Linux package, so for now I've disabled building the package for i686:

sgn, 2022-09-18:
```diff
--- a/dfu/device/stm32wb55/infotable.h
+++ b/dfu/device/stm32wb55/infotable.h
@@ -52,6 +52,7 @@ struct FUSDeviceInfoTable
     uint32_t  reserved2;
     uint64_t  UID64;
     uint16_t  deviceID;
+    uint32_t  padding_ /* padding for 32 bits build */;
 };
 
 struct DeviceInfoTable {
```
This fixes build for 1.1.3, I didn't try with master.

gsurkov, 2022-09-19:
@RunningDroid Unfortunately, qFlipper was never tested on 32-bit hardware. Please use **sgn**'s patch, test it with your package and file a pull request in case of a satisfactory result. 
There should be no difference between 1.1.3 and dev branches because there were no changes to those files.

commit 8b927e4b (2022-11-07): Fix padding for 32 bit builds (Fixes #120) (#131)
closed 2022-09-19 by gsurkov