flipperdevices/qFlipper #123: Unable to drag and drop files and folders into file manager in qFlipper 0.66.1

jmgelman, 2022-09-21:
**Describe the bug**
Drag and drop files and folders into the File Manager in qFlipper 0.66.1

**To Reproduce**
Steps to reproduce the behavior:
1. Open qFlipper.
2. Navigate to File Manager.
3. Open Windows Explorer.
4. Attempt to drag and drop a file or folder from Windows Explorer.

**Expected behavior**
Dropping file or folder should upload the file to the Flipper SD card.

**Actual behavior**
Cursor changes to "not-allowed" and cannot drop file or folder.

**Additional troubleshooting**
Issue did not appear in previous version of qFlipper.

**Screenshot**
[image]

gsurkov, 2022-09-22:
I was not able to reproduce it on Windows 10.
Which exact version of Windows 10 are you using? (you can find it out right at the beginning of qFlipper's log.)
Does the path to the file you are attempting to drag contain any peculiar symbols?
Does the problem persist if you restart qFlipper? If you restart Windows? 
There allegedly was a case of "not allowed" cursor if the user tried to drag a file into the root directory (which is not allowed), and then attempted to drag it into another folder. Is it what you did? 
Does your system have multiple displays? If so, does moving qFlipper onto the other display do anything?

jmgelman, 2022-09-22:
@gsurkov Thank you, that appears to be the cause. I chose the "Run qFlipper now" option after the installer completed. I was not aware that drag-and-drop was disabled when the app is run as an administrator. Is that by design?

gsurkov, 2022-09-22:
Windows disables drag-and-drop from regular programs to privileged ones as a security measure (only learned this today), so qFlipper has nothing to do with it.

Closing this issue, but will keep in mind that the installer needs be fixed so that it would launch qFlipper without admin privileges.

closed 2022-09-22 by gsurkov