flipperdevices/qFlipper #124: Opening the file explorer under windows kills qFlipper.

SplitPixl, 2022-09-24:
**Describe the bug**
When doing anything that calls the Windows File Explorer (eg: installing firmware from file, backup and restoring internal flash), qFlipper dies.

**To Reproduce**
Steps to reproduce the behavior:
1. Open qFlipper
2. Click "Install From File"
3. Observe as qFlipper disappears.

**Expected behavior**
I should be able to pick a file to send to my device.

**Actual behavior**
qFlipper vanishes before my eyes.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? **Yes.**
2. Does reconnecting/rebooting Flipper solve it? **No.**

**Additional context**
The workaround I'm currently using is opening a command prompt in my qFlipper install folder, and running `qFlipper-cli.exe firmware <path/to/firmware.dfu>`

**Logs**
```
670 [APP] qFlipper version 1.2.0 commit 49a2c9b4 2022-09-19T06:02:27
671 [APP] OS info: Windows 10 Version 2009 10 10.0.22000 Qt 5.15.2
1014 [DBG] Trying serial port  at \\.\COM7
1014 [DBG] Trying serial port  at \\.\COM6
1014 [DBG] Trying serial port FLIP_MYFLIPPER at \\.\COM8
1014 [DBG] Using  serial port FLIP_MYFLIPPER at \\.\COM8
1015 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
1015 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
1021 [RPC] Starting RPC session...
1067 [RPC] RPC session started successfully.
1077 [RPC] (1) System Device Info START
1185 [RPC] (1) System Device Info SUCCESS
1185 [RPC] (2) Storage Info @/ext START
1237 [RPC] (2) Storage Info @/ext SUCCESS
1237 [RPC] (3) Storage Stat @/ext/Manifest START
1294 [RPC] (3) Storage Stat @/ext/Manifest SUCCESS
1295 [RPC] (4) System Get DateTime START
1346 [DBG] Flipper time skew is -1098 milliseconds
1346 [RPC] (4) System Get DateTime SUCCESS
1346 [RPC] (5) System Set DateTime START
1398 [RPC] (5) System Set DateTime SUCCESS
1398 [RPC] Stopping RPC session...
1400 [RPC] Attempting to unload protobuf plugin...
1400 [RPC] Unloaded protobuf plugin.
1400 [RPC] RPC session stopped successfully.
1400 [REG] Registering the device
1400 [DEV] Version: 0.67.3 commit: bc9aa6545 radio: 1.13.3
1400 [BKD] Current device changed to "MyFlipper"
1501 [RPC] Starting RPC session...
1543 [RPC] RPC session started successfully.
1543 [RPC] (1) Gui Start ScreenStream START
1601 [RPC] (1) Gui Start ScreenStream SUCCESS
```

ase1590, 2022-09-24:
Adding in here, I've seen this issue with about 6 or so people on windows in the discord channel, so its certainly not an isolated occurrence. in *some* cases downgrading from a version earlier than 1.2.0 works for these individuals afflicted by this.

Noisypegasus, 2022-09-24:
Same issue on both default, RC, Dev, and RM (current loadout)
 Captured after I loaded RM on an earlier release of qFlipper (which worked). Then updated to current qFlipper to recreate the result and sent the current error log.
 
 518 [APP] qFlipper version 1.2.0 commit 49a2c9b4 2022-09-19T08:02:27
519 [APP] OS info: Windows 10 Version 2009 10 10.0.19043 Qt 5.15.2
645 [DBG] Trying serial port 00E04C239987 at \\.\COM8
645 [DBG] Trying serial port  at \\.\COM1
645 [DBG] Trying serial port FLIP_LEPT1P at \\.\COM9
645 [DBG] Using  serial port FLIP_LEPT1P at \\.\COM9
653 [RPC] Starting RPC session...
707 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
708 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
848 [RPC] RPC session started successfully.
849 [RPC] (1) System Device Info START
956 [RPC] (1) System Device Info SUCCESS
956 [RPC] (2) Storage Info @/ext START
1013 [RPC] (2) Storage Info @/ext SUCCESS
1013 [RPC] (3) Storage Stat @/ext/Manifest START
1081 [RPC] (3) Storage Stat @/ext/Manifest SUCCESS
1081 [RPC] (4) System Get DateTime START
1146 [DBG] Flipper time skew is -1482 milliseconds
1146 [RPC] (4) System Get DateTime SUCCESS
1146 [RPC] (5) System Set DateTime START
1213 [RPC] (5) System Set DateTime SUCCESS
1213 [RPC] Stopping RPC session...
1235 [RPC] Attempting to unload protobuf plugin...
1235 [RPC] Unloaded protobuf plugin.
1235 [RPC] RPC session stopped successfully.
1235 [REG] Registering the device
1235 [DEV] Version: 0.67.3 commit: d9ac18f3 radio: 1.13.3
1235 [BKD] Current device changed to "Lept1p"
1390 [RPC] Starting RPC session...
1566 [RPC] RPC session started successfully.
1566 [RPC] (1) Gui Start ScreenStream START
1629 [RPC] (1) Gui Start ScreenStream SUCCESS
7891 [] QWindowsNativeFileDialogBase::shellItem: SHCreateItemFromParsingName(file:///F:/Desktop/0.66.3-0921-RMv2)) failed (The system cannot find the file specified.)

SplitPixl, 2022-09-24:
I've done some digging and found some more info. This issue was (for me) being caused by using a file on a removable disk, then removing the disk. qFlipper tries to open the previous path, which doesn't exist anymore.

This issue could be resolved by catching an error here, or adding logic to check if the path exists before trying to open it.
https://github.com/flipperdevices/qFlipper/blob/49a2c9b42c8e5c130272a829b1866f1187e3b5f9/application/components/FileManager.qml#L241

sundhaug92, 2022-09-24:
Maybe try to open, and if it doesn't work provide a warning and then continue on?

gsurkov, 2022-09-25:
Gonna check it myself soon. Verifying that the saved path is openable sounds reasonable, it's strange that Qt or the OS do not do it under the hood.
If it's really caused by removable drives, then editing/removing the `LastFolderUrl` in qFlipper's config file (should be found somewhere in the AppData folder) will stop qFlipper from crashing.

lastCoyotes, 2022-09-28:
Not a fix to the issue, but rather a quick fix until there's logic patched in to check if the file location exists.

Under AppData > Local > qFlipper
(of if you open logs on qFlipper, right-click and select "Browse all logs...")
open the latest text log file after qFlipper crashed trying to open the File Browser.

You should look for a line with the error:
22718 [] QWindowsNativeFileDialogBase::shellItem: SHCreateItemFromParsingName(file:///<file path here>)) failed (The system cannot find the path specified.)

Create the chain of folders again following the folder names it logged, and then you should be able to access the file browser within qFlipper again. Be sure to close the File Browser opened by qFlipper on a folder you're sure to not delete.

gsurkov, 2022-10-03:
Fixed in 1.2.1.

closed 2022-10-03 by gsurkov