flipperdevices/qFlipper #126: Publish Flatpak on Flathub

kowalski7cc, 2022-09-27:
Flathub is an application repository which uses the Flatpak distribution method.

Flatpak allows applications to be compatible with almost all linux distros because they are compiled and run in a container-like environment.
You can know more about flatpak and flathub at https://flatpak.org/ and https://flathub.org/

I made a working flatpak at the repository
https://github.com/kowalski7cc/flathub/tree/one.flipperzero.qFlipper

There are a couple of issues for now:
- You need to manually run the command in the previous message to manually install udev rules, wich kinda defeats sandbox and it's not user friendly `flatpak run --talk-name=org.freedesktop.Flatpak --command=setup_rules one.flipperzero.qFlipper`
- File picker dialog has some issue in opening file when sandbox not allows every path, but save dialog doesn't have the same issue. If you add access to home this is fixed but again, kinda defeats sandboxing
- One improvement also could be found in disabling auto update and to rely on flatpak update system. But this wold be a patch to source (that could be also upstreamed, so just chek if running in flatpak then disable auto updater)

Chadnelson31, 2022-09-27:
This would be great. Some other added benefits of publishing a flatpak version:

1. Allows qFlipper to show up under Gnome Software and Discover app stores for distributions that have the correct appstream packages (most mainline distributions). This makes obtaining qFlipper as easy as opening an App Store and searching for it.

2. Allows distributions with immutable filesystems and nonstandard file hierarchies to install the package easily (Fedora Silverblue, SteamOS, Guix, NixOS, etc..)

3. Integrates qFlipper cleanly with most desktop environments such as generating the correct .desktop files. This is probably the main downside of the appimage (outside of the distributions where appimages aren't supported).

gsurkov, 2022-09-28:
@kowalski7cc  It is possible to disable the built-in application update feature by passing `DEFINES+=DISABLE_APPLICATION_UPDATES` to the `qmake` call when building qFlipper. It was specifically aimed at packaging support.

Having a flatpak would be nice, but this is on the community. We don't have the resources to support every possible package system or distro. More so, flatpaks have the aforementioned limitations (e.g. no reasonable way to install system files).

I would gladly help anyone who would take on this task and implement relevant changes to qFlipper's code, when necessary.

closed 2022-09-28 by gsurkov

kowalski7cc, 2022-09-28:
@gsurkov If for flipperdevices is ok, I can maintain it as I am already maintaining Space Cadet on flathub.

gsurkov, 2022-09-30:
@kowalski7cc Thanks a lot!

razzeee, 2022-10-02:
Hrm, did the code change? When I tried to package it, it did not redetect after flashing

https://github.com/flipperdevices/qFlipper/issues/95#issuecomment-1204565526

kowalski7cc, 2022-10-02:
What do you mean? If you mean the permission issue "read more" it has been changed to point to a future wiki to guide the user to set up the udev rules instead of the docs home page. But the wiki would be hosted on the future Flathub repository that doesn't exists yet. You can find the code changed in the patch file in the pull request.

razzeee, 2022-10-02:
Like I pointed out in that issue and just tested with your flatpak:

- if you first start the flatpak and then connect the flipper, the app will never know it's connected
- similar, if you update the firmware, the device restarts and qflipper tries to wait for that, as it never get's the reconnect event, it will wait forever

Also your implementations seems to not be startable from the terminal?

kowalski7cc, 2022-10-02:
Oh yes, sadly that's a issue on flatpak's side.
For the terminal it seem's I need to look in why `qFlipper-cli` can't load `libflipperproto0.so`