flipperdevices/qFlipper #127: Crashes when attempting to install from file etc

JetLynx, 2022-09-28:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**in Windows 10, qFlipper crashes when I try to install firmware from file, Backup, or Restore. Attempting to factory reset using Erase function also stays on "Performing Factory Reset" screen and never fails nor succeeds**
What were you trying to do? Install custom firmware, specifically RogueMaster. At this point I can't do any of the above functions.

**To Reproduce**
Steps to reproduce the behavior:
1. Run qFlipper, either normally or as Administrator
2. Click on install from file, Backup, or Restore
3. See qFlipper crash immediately

**Expected behavior**
Firmware flashing process to run and complete. Or any of the above functions to run without crashing.

**Actual behavior**
qFlipper crashes without any error messages or dialog windows.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? Yes
2. Does reconnecting/rebooting Flipper solve it? No
3. Does re-installing solve the issue? No
4. Does running as Administrator solve it? No
5. Does restarting the computer solve it? No
6. Does formatting the SD Card and performing the factory update and database install solve it? No

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
1. Click the "LOGS" button to open the log window.
2. Right click on the log window.
7. In the context menu, select "Browse all logs...".
8. A folder containing several log files should open. 
9. Find the most recent file, this is it!

**Warning!** The log will contain your Flipper's unique name. If you wish not to disclose it, search/replace
all its occurrences before attaching the file.
[qFlipper-20220927-204941.txt](https://github.com/flipperdevices/qFlipper/files/9661337/qFlipper-20220927-204941.txt)

lastCoyotes, 2022-09-28:
Quickfix until a solution patches the logic, but for anyone else who has this issue it's due to qFlipper opening up the File Browser to a location that doesn't exist if it saved that location as the last location used.

https://github.com/flipperdevices/qFlipper/issues/124#issuecomment-1260480394

JetLynx, 2022-09-28:
> Quickfix until a solution patches the logic, but for anyone else who has this issue it's due to qFlipper opening up the File Browser to a location that doesn't exist if it saved that location as the last location used.
> 
> [#124 (comment)](https://github.com/flipperdevices/qFlipper/issues/124#issuecomment-1260480394)

I recreated the directory that's listed at the end of the log and now all issues are resolved.

gsurkov, 2022-09-28:
Duplicate of #124. The fix will be released as soon as we fix our build system.

closed 2022-09-28 by gsurkov