flipperdevices/qFlipper #133: fails to copy bunch of files

cherepanov, 2022-11-19:
qFlipper 1.2.1 on Ubuntu 22.10 fails to copy bunch of flies. Probably out of file descriptors

1) Clone this repos
```
https://github.com/logickworkshop/Flipper-IRDB
https://github.com/sasiplavnik/Flipper-IRDB
```

2) Drag these folders to sdcard view in qFlipper

3) Soon you'll get an error

```
94561 [RPC] (227) Storage MkDir @/ext/infrared/Flipper-IRDB-irplus/STARSAT START
94561 [RPC] Serial connection was lost.
94561 [RPC] Stopping RPC session...
94561 [UTL] Upload Files @Rer0g4x ERROR: RPC session was stopped with operations still running
94565 [RPC] (227) Storage MkDir @/ext/infrared/Flipper-IRDB-irplus/STARSAT ERROR: RPC session was stopped with operations still running
94676 [RPC] Attempting to unload protobuf plugin...
94676 [RPC] Unloaded protobuf plugin.
94677 [RPC] RPC session stopped successfully.
94677 [BKD] Current operation finished with error: "Resource temporarily unavailable"
94678 [RPC] Stopping RPC session...
94678 [RPC] RPC session stopped successfully.
94680 [BKD] Current operation finished with error: "Resource temporarily unavailable"
```

[image]

gsurkov, 2022-11-25:
There are several things that cause this.
First things first, it's really possible to make qFlipper run out of file descriptors as it (wrongly) frees them only after the current batch has fully completed. This has been fixed in fc0003eeb37b4a312393d7092917f4eadc5bd76d.
Next, the file name filtering rules have been adjusted in the Flipper Zero [firmware](https://github.com/flipperdevices/flipperzero-firmware/pull/2047) as some files in the above repositories did not pass the checks (albeit having perfectly valid names). The fix will be available once the PR gets merged.
Third, qFlipper now will try to fix forbidden characters in the file names (which makes more sense to me rather than cancelling a transfer of >9000 files).

As a general remark, copying such a large number of files via qFlipper does not seem like a reasonable thing to do anyway. It was fixed more for the neatness' sake, rather than being actually useful. The amount of time it takes to transfer all those files is absolutely obscene and it would be much faster to do that using a card reader.

One more thing: one should not copy git repositories directly as they contain a hidden `.git` directory packed full of various files that are useless for the target application, which will surely cause all sorts of unnecessary waiting. Always delete it before copying the whole folder.

Will close the issue once the aforementioned PR gets merged.

skotopes, 2022-11-28:
Fixed in dev, will arrive in next release.

closed 2022-11-28 by skotopes