flipperdevices/qFlipper #140: Cannot connect to update server

schemil053, 2022-12-10:
[image]
You can see the error on the Screenshot.

schemil053, 2022-12-10:
[UPD] Failed to parse update information: ChannelInfo: Expected an array of versions
[UPD] Failed to parse update information: ChannelInfo: Expected an array of versions
is in the Log

CrookedBIOS, 2022-12-10:
Same here on Fedora 36
Logfile:

```
279 [APP] qFlipper version 1.2.1 commit 3ec0fd9d 2022-10-03T17:15:41
279 [APP] OS info: Fedora Linux 36 (KDE Plasma) 36 6.0.10-200.fc36.x86_64 Qt 5.15.2
296 [REG] Detected new device: VID_0x483:PID_0x5740
378 [DBG] Trying serial port flip_xxxxxx at /dev/ttyACM0
378 [DBG] Using  serial port flip_xxxxxx at /dev/ttyACM0
378 [RPC] Starting RPC session...
426 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
444 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
444 [UPD] Failed to parse update information: ChannelInfo: Expected an array of versions
473 [RPC] RPC session started successfully.
473 [RPC] (1) System Device Info START
581 [RPC] (1) System Device Info SUCCESS
581 [RPC] (2) Storage Info @/ext START
633 [RPC] (2) Storage Info @/ext SUCCESS
633 [RPC] (3) Storage Stat @/ext/Manifest START
691 [RPC] (3) Storage Stat @/ext/Manifest SUCCESS
692 [RPC] (4) System Get DateTime START
743 [DBG] Flipper time skew is -1257 milliseconds
743 [RPC] (4) System Get DateTime SUCCESS
743 [RPC] (5) System Set DateTime START
795 [RPC] (5) System Set DateTime SUCCESS
795 [RPC] Stopping RPC session...
797 [RPC] Attempting to unload protobuf plugin...
797 [RPC] Unloaded protobuf plugin.
797 [RPC] RPC session stopped successfully.
797 [REG] Registering the device
797 [DEV] Version: 0.66.1 commit: 55b46324 radio: 1.13.3
797 [BKD] Current device changed to "Lys10n"
898 [RPC] Starting RPC session...
990 [RPC] RPC session started successfully.
990 [RPC] (1) Gui Start ScreenStream START
1045 [RPC] (1) Gui Start ScreenStream SUCCESS
2248 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
2248 [UPD] Failed to parse update information: ChannelInfo: Expected an array of versions
```

CrookedBIOS, 2022-12-10:
Works now. Can update.

Seems like the issue was that `"versions": null` was defined in the development channel.
URL with JSON now contains a valid version in development channel, and I am able to update the Flipper Zero firmware using qFlipper

fritolays, 2022-12-11:
Once again a null in the dev channel and cant update....

fritolays, 2022-12-11:
Hmm, must be while the system is uploading a new build or during a build process.
I dont know, but yes same as above, works now and dev channel reads normally.

gsurkov, 2022-12-12:
It is a problem with the web backend, not with qFlipper. It is being worked on.

closed 2022-12-12 by gsurkov