flipperdevices/qFlipper #144: Support for flashing custom named FW

FalsePhilosopher, 2022-12-24:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
Building the FW with a custom name does not work in qflipper. It works with web flasher and SD update but not qflipper.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash FW with custom name with qflipper
2. See error

**Expected behavior**
It should flash but gives an error instead.

**Actual behavior**
A clear and concise description of what happened instead.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times?
2. Does reconnecting/rebooting Flipper solve it?

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

**Additional context**
You are free to add any other context about the problem you deem helpful here.

## IMPORTANT - Getting the full log
Normally, the application log is shown upon pressing the "LOGS" button in the bottom left corner of the window. 
Despite it having some value already, there is a more detailed version saved each time qFlipper is run.
In order to access it, follow these steps:
1. Click the "LOGS" button to open the log window.
2. Right click on the log window.
3. In the context menu, select "Browse all logs...".
4. A folder containing several log files should open. 
5. Find the most recent file, this is it!

**Warning!** The log will contain your Flipper's unique name. If you wish not to disclose it, search/replace
all its occurrences before attaching the file.

gsurkov, 2022-12-25:
What do you mean by "custom name"? Could you please give an example or a more detailed explanation of how to reproduce it?

FalsePhilosopher, 2022-12-25:
> What do you mean by "custom name"? Could you please give an example or a more detailed explanation of how to reproduce it?

Build the FW with this option
CUSTOM_FLIPPER_NAME=Name ./fbt COMPACT=1 DEBUG=0 updater_package

gsurkov, 2022-12-26:
Changing Flipper's name is not a part of the official Flipper Zero firmware and is therefore unsupported. 
QFlipper relies on the device name in order to keep track of it during the update process and must remain the same all the way through.

closed 2022-12-26 by gsurkov