flipperdevices/qFlipper #150: Project ERROR: Unknown module(s) in QT: core5compat

stappersg, 2022-12-31:
Hi,

Here a hobby programmer without experience with QT.

What I would like to do, is building qFlipper natively on a recent Debian system.
(FWIW: compiling with Docker (that uses Ubuntu 18.04) is succesfull)

I have installed the `qt6-base-dev` package as suggested / hinted by the `README.md`.
Continueing with that guide:
```text
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ qmake ../qFlipper.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
bash: qmake: command not found
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ qmake6 ../qFlipper.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ make qmake_all        
cd 3rdparty/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/3rdparty/3rdparty.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
cd 3rdparty/ && make -f Makefile qmake_all
make[1]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
make[1]: Nothing to be done for 'qmake_all'.
make[1]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
cd dfu/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/dfu/dfu.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
Project ERROR: Unknown module(s) in QT: core5compat
make: *** [Makefile:121: sub-dfu-qmake_all] Error 3
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ dpkg -l 'q*dev*'      # to show that QT 6 development libraries are available
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name               Version          Architecture Description
+++-==================-================-============-=================================
ii  qt6-base-dev:amd64 6.4.2+dfsg~rc1-3 amd64        Qt 6 base development files
ii  qt6-base-dev-tools 6.4.2+dfsg~rc1-3 amd64        Qt 6 base development programs
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ 
```

What to do to get `qFlipper` compiled natively on are recent Debian system?

stappersg, 2022-12-31:
For what it is worth:
```text
$ git diff
diff --git a/qflipper_common.pri b/qflipper_common.pri
index 6266bad..9ec5e93 100644
--- a/qflipper_common.pri
+++ b/qflipper_common.pri
@@ -1,6 +1,6 @@
 NAME = qFlipper
 
-equals(QT_MAJOR_VERSION, 6): QT += core5compat
+equals(QT_MAJOR_VERSION, 6): QT += core6compat
 
 unix:!macx {
     DEFINES += USB_BACKEND_LIBUSB
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ make qmake_all
cd 3rdparty/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/3rdparty/3rdparty.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
cd 3rdparty/ && make -f Makefile qmake_all
make[1]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
make[1]: Nothing to be done for 'qmake_all'.
make[1]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
cd dfu/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/dfu/dfu.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
Project ERROR: Unknown module(s) in QT: core6compat
make: *** [Makefile:121: sub-dfu-qmake_all] Error 3
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ 
```

stappersg, 2022-12-31:
Seems that package `libqt6core5compat6-dev` is also needed.
```text
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ apt-file search core5compat
libqt6core5compat6: /usr/share/doc/libqt6core5compat6/changelog.Debian.amd64.gz
libqt6core5compat6: /usr/share/doc/libqt6core5compat6/changelog.Debian.gz
libqt6core5compat6: /usr/share/doc/libqt6core5compat6/copyright
libqt6core5compat6-dev: /usr/include/x86_64-linux-gnu/qt6/QtCore5Compat/qtcore5compat-config.h
libqt6core5compat6-dev: /usr/include/x86_64-linux-gnu/qt6/QtCore5Compat/qtcore5compatversion.h
libqt6core5compat6-dev: /usr/lib/x86_64-linux-gnu/metatypes/qt6core5compat_none_metatypes.json
libqt6core5compat6-dev: /usr/lib/x86_64-linux-gnu/qt6/mkspecs/modules/qt_lib_core5compat.pri
libqt6core5compat6-dev: /usr/share/doc/libqt6core5compat6-dev/changelog.Debian.amd64.gz
libqt6core5compat6-dev: /usr/share/doc/libqt6core5compat6-dev/changelog.Debian.gz
libqt6core5compat6-dev: /usr/share/doc/libqt6core5compat6-dev/copyright
qt6-5compat-dev: /usr/include/x86_64-linux-gnu/qt6/QtCore5Compat/qtcore5compat-config.h
qt6-5compat-dev: /usr/include/x86_64-linux-gnu/qt6/QtCore5Compat/qtcore5compatversion.h
qt6-5compat-dev: /usr/lib/x86_64-linux-gnu/metatypes/qt6core5compat_none_metatypes.json
qt6-5compat-dev: /usr/lib/x86_64-linux-gnu/qt6/mkspecs/modules/qt_lib_core5compat.pri
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ sudo apt install qt6-5compat-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 qt6-5compat-dev : Depends: libqt6core5compat6 (= 6.4.2~rc1-1) but 6.3.1-2+b1 is to be installed
E: Unable to correct problems, you have held broken packages.
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ sudo apt install libqt6core5compat6-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libqt6core5compat6
The following NEW packages will be installed:
  libqt6core5compat6 libqt6core5compat6-dev
0 upgraded, 2 newly installed, 0 to remove and 2069 not upgraded.
Need to get 156 kB of archives.
After this operation, 609 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://deb.debian.org/debian unstable/main amd64 libqt6core5compat6 amd64 6.3.1-2+b1 [132 kB]
Get:2 http://deb.debian.org/debian unstable/main amd64 libqt6core5compat6-dev amd64 6.3.1-2+b1 [23.4 kB]
Fetched 156 kB in 0s (314 kB/s)              
Selecting previously unselected package libqt6core5compat6:amd64.
(Reading database ... 600010 files and directories currently installed.)
Preparing to unpack .../libqt6core5compat6_6.3.1-2+b1_amd64.deb ...
Unpacking libqt6core5compat6:amd64 (6.3.1-2+b1) ...
Selecting previously unselected package libqt6core5compat6-dev:amd64.
Preparing to unpack .../libqt6core5compat6-dev_6.3.1-2+b1_amd64.deb ...
Unpacking libqt6core5compat6-dev:amd64 (6.3.1-2+b1) ...
Setting up libqt6core5compat6:amd64 (6.3.1-2+b1) ...
Setting up libqt6core5compat6-dev:amd64 (6.3.1-2+b1) ...
Processing triggers for libc-bin (2.36-6) ...
======  How can you help?  (doc: https://wiki.debian.org/how-can-i-help ) ======

New packages removed from Debian 'testing' (the maintainer might need help):
 - libruby3.0 - https://tracker.debian.org/pkg/ruby3.0
 - ruby3.0 - https://tracker.debian.org/pkg/ruby3.0

-----  Show old opportunities as well as new ones: how-can-i-help --old  -----
Scanning processes...                                                                                                                              
Scanning candidates...                                                                                                                             
Scanning processor microcode...                                                                                                                    
Scanning linux images...                                                                                                                           

[... 16 lines cut ...]
$ git restore ../qflipper_common.pri
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ make qmake_all
cd 3rdparty/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/3rdparty/3rdparty.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
cd 3rdparty/ && make -f Makefile qmake_all
make[1]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
make[1]: Nothing to be done for 'qmake_all'.
make[1]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
cd dfu/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/dfu/dfu.pro PREFIX=/path/to/install/dir -spec linux-g++ CONFIG+=qtquickcompiler
Failure to open file: /usr/src/github/flipperdevices/qFlipper/build/dfu/Makefile
Unable to generate makefile for: /usr/src/github/flipperdevices/qFlipper/dfu/dfu.pro
make: *** [Makefile:121: sub-dfu-qmake_all] Error 5
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ 
```

For now I leave this issue open as reference for others that might encounter
```text
Project ERROR: Unknown module(s) in QT: core5compat
```

I think that sustainable solution is updating `README.md` section _Linux Standalone build_.
> Build requirements:
> - Qt5 >= 5.15.0 or Qt6 >= 6.3.0
> - libusb >= 1.0.16

gsurkov, 2023-01-01:
When Qt is being packaged by a Linux distribution, it is usually separated to several modules which implement a certain functionality. These modules correspond to those in .pro files (lines with `QT += ....` statements).

Although not explicitly stated in the README.md file, it is implied that one must install all of the required packages. Since they are called a bit differently in each distribution, it's not possible to write the exact names, although it would be useful to list the required Qt modules there.

The easiest way to find it out would be to look [here](https://github.com/flipperdevices/qFlipper/blob/dev/docker/Dockerfile), find the corresponding Qt packages in your distribution and install them via the system package manager. This should fix these errors

Please bear in mind that some Linux distributions may have outdated or incomplete packages, which can cause errors on its own (hence it's best to just use a container).

closed 2023-01-01 by gsurkov

sur5r, 2023-01-11:
@stappersg FWIW, qflipper is available in Debian sid and bookworm.

stappersg, 2023-01-11:
> @stappersg FWIW, qflipper is available in Debian sid and bookworm.

Indeed, [it is](https://packages.debian.org/search?keywords=qflipper). Thanks for telling.