flipperdevices/qFlipper #151: Project ERROR: Unknown module(s) in QT: serialport

stappersg, 2022-12-31:
Hi, here the reporter of #150 

Trying to do a _Linux standalone build_.

Installed are some QT6 development libraries and `libusb-dev`.

But I get: `Project ERROR: Unknown module(s) in QT: serialport`, in context:
```text
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ make qmake_all
cd 3rdparty/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/3rdparty/3rdparty.pro PREFIX=/usr/local/bin -spec linux-g++ CONFIG+=qtquickcompiler
cd 3rdparty/ && make -f Makefile qmake_all
make[1]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
make[1]: Nothing to be done for 'qmake_all'.
make[1]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/3rdparty'
cd dfu/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/dfu/dfu.pro PREFIX=/usr/local/bin -spec linux-g++ CONFIG+=qtquickcompiler
cd dfu/ && make -f Makefile qmake_all
make[1]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/dfu'
make[1]: Nothing to be done for 'qmake_all'.
make[1]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/dfu'
cd plugins/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/plugins/plugins.pro PREFIX=/usr/local/bin -spec linux-g++ CONFIG+=qtquickcompiler
cd plugins/ && make -f Makefile qmake_all
make[1]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/plugins'
cd flipperproto0/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/plugins/flipperproto0/flipperproto0.pro PREFIX=/usr/local/bin -spec linux-g++ CONFIG+=qtquickcompiler
cd flipperproto0/ && make -f Makefile qmake_all
make[2]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/plugins/flipperproto0'
make[2]: Nothing to be done for 'qmake_all'.
make[2]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/plugins/flipperproto0'
cd protobufinterface/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/plugins/protobufinterface/protobufinterface.pro PREFIX=/usr/local/bin -spec linux-g++ CONFIG+=qtquickcompiler
cd protobufinterface/ && make -f Makefile qmake_all
make[2]: Entering directory '/usr/src/github/flipperdevices/qFlipper/build/plugins/protobufinterface'
Some of the required modules (false) are not available.
Skipped.
make[2]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/plugins/protobufinterface'
make[1]: Leaving directory '/usr/src/github/flipperdevices/qFlipper/build/plugins'
cd backend/ && /usr/bin/qmake6 -o Makefile /usr/src/github/flipperdevices/qFlipper/backend/backend.pro PREFIX=/usr/local/bin -spec linux-g++ CONFIG+=qtquickcompiler
Project ERROR: Unknown module(s) in QT: serialport
make: *** [Makefile:95: sub-backend-qmake_all] Error 3
stappers@messer:/usr/src/github/flipperdevices/qFlipper/build
$ 
```

Which development library is missing?

P.S.

stappersg, 2022-12-31:
Short:
```text
sudo apt install qt6-serialport-dev
```

Long:
In file `../docker/Dockerfile` is a list of _debian packages_ to be installed. Seeing a `serialport` package inspired me to do `apt search serialport | grep qt` for finding an eqivalent.

stappersg, 2022-12-31:
Issue left open by me because I deem that [`- Qt5 >= 5.15.0 or Qt6 >= 6.3.0`](https://github.com/flipperdevices/qFlipper/blob/dev/README.md?plain=1#L58) is way too terse.

gsurkov, 2023-01-01:
Gave an explanation in #150, but you seem to have figured it out on your own.
Best I can do is add a list of required Qt modules in Readme.md (which I did), the exact package names are distro-dependent and will still need to be found by the user.

closed 2023-01-01 by gsurkov