flipperdevices/qFlipper #155: USBDeviceInfo::isComplete() breaks windows USB detection

L4ys, 2023-01-05:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
The new commit 78613f6a3ce6acc24afdf47c9be7b2f3495c5b5c breaks USB detection on Windows 

**To Reproduce**
Connect Flipper Zero to USB port and run dev build qFlipper on Windows

**Expected behavior**
Flipper Zero connected to qFlipper

**Actual behavior**
Flipper Zero not detected or qFlipper shows the device is not supported.

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
```
1572 [APP] qFlipper version 1.2.2 commit 84566a50 2023-01-02T01:59:28
1573 [APP] OS info: Windows 10 Version 2009 10 10.0.19045 Qt 5.15.2
2068 [UPD] Fetched update information from https://update.flipperzero.one/qFlipper/directory.json
2072 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
```

Log from qFlipperZero-cli:
```
>qFlipper-cli.exe
REG: Incomplete device info: VID_0x483:PID_0x5740
3 [CLI] Performing full firmware update...
3 [CLI] Waiting for devices...
```

**Additional context**

The actual problem is that the new `USBDeviceInfo::isComplete()` now checks `m_manufacturer` and `m_productDescription`:
https://github.com/flipperdevices/qFlipper/blob/78613f6a3ce6acc24afdf47c9be7b2f3495c5b5c/dfu/usbdeviceinfo.cpp#L10-L12

and these two fields are not filled in the windows usb backend before `devicePluggedIn` emited:
https://github.com/flipperdevices/qFlipper/blob/84566a5075f42da75292feced0977fddb23c11fa/dfu/win32/usbdevicedetector.cpp#L165-L177

we should get these two fields for windows backend or not check them.

closed 2023-03-30 by gsurkov

arthurlutz, 2023-04-03:
Am getting 

```
986 [USB] Failed to open device: Access denied (insufficient permissions)
1006 [REG] Incomplete device info: VID_0x483:PID_0x5740
```

which seems very similar. On the 1.3.0 version. @gsurkov in which version is the bug fixed ? Can you point to a pull request or a commit ?

gsurkov, 2023-04-03:
@arthurlutz it was fixed in fdf5997, long before 1.3.0.
Your problem seems unrelated, so please open another issue.
(But before that, if perchance you are using Linux, do not forget to install the [device access permissions](https://github.com/flipperdevices/qFlipper#linux-1)).

arthurlutz, 2023-04-03:
Indeed, sorry for the noise. I ended up fixing the permission problem.