flipperdevices/qFlipper #156: qFlipper does not recognize an updated firmware bundle

hedger, 2023-01-11:
**Describe the bug**
qFlipper does not recognize an updated firmware bundle.

**To Reproduce**
Steps to reproduce the behavior:

 See https://github.com/flipperdevices/flipperzero-firmware/issues/1881

**Expected behavior**
An update package is uploaded and installed.

**Actual behavior**
Update package isn't installed. 

More info in the thread above. @[ankris812](https://github.com/ankris812)

[a.txt](https://github.com/flipperdevices/qFlipper/files/10393641/a.txt)

gsurkov, 2023-01-11:
Looks like a very minor bug in a special occasion, but it won't hurt to add CRC checks for all existing files to see whether they should be reuploaded or not.

gsurkov, 2023-01-23:
Now qFlipper checks the md5 sums of the files on the SD card, if it finds any. It will also reupload any missing files.

closed 2023-01-23 by gsurkov