flipperdevices/qFlipper #158: Building on MacOS x86_64

TQMatvey, 2023-01-12:
**Describe the bug**
Building qFlipper on Mac OS Ventura and earlier with X86_64 architecture

**To Reproduce**
Steps to reproduce the behavior:
1. Get `qt, libusb` packages from brew
2. `./build_mac.sh`
3. `Project ERROR: libusb-1.0 development package not found`

**Expected behavior**
i should be able to download flipper's libusb with `brew install libusb_universal`

**Actual behavior**
running `brew install libusb_universal` throws `libusb_universal: The arm64 architecture is **required** for this software`

**Additional troubleshooting**
I tried googling, installing other libusb packages with brew, pip3 

**The report *must* also include at least one of the following:**
[image]
[image]

gsurkov, 2023-01-12:
This build process was designed for our automatic build server, which is running on an ARM mac, so it wasn't tested for x86_64 much.

We'll try to adapt it so it would work on any architecture, but it may take some time. (Pull requests will be greatly appreciated!)

Qt and libusb must not necessarily be installed from the flipper tap, however, it's possible to use the regular brew versions, depending on your needs (should be sufficient for development purposes, for example).

sprrw, 2023-02-27:
It would be great to have `qflipper` formula/cask in brew though. Even better, also have `fbt` as a brew formula.