flipperdevices/qFlipper #160: Keyboard passthru for text entry via qFlipper

Gho57X90, 2023-01-29:
I would like to request the feature for users to use their keyboard to input text into the text entry fields in apps for the Flipper Zero via qFlipper.

In general, it's slow to input text into the Flipper Zero as-is, which is not a big deal in most casual use-cases. _But_, when you want to save several remotes/NFC tags/etc. en masse _with_ proper names, or even input long strings of text in apps such as the [Flipper Authenticator](https://github.com/akopachov/flipper-zero_authenticator) app by @akopachov, it becomes incredibly tedious with the very basic arrow controls currently available.

This is also a big deal for accessibility reasons when a Flipper app does not have a config file that can be easily edited in bulk by a user who does not have a full understanding of how it works. Again, pointing to the Flipper Authenticator app as an example; of which generates a config file that, at least to the uneducated/inexperienced, cannot simply be edited in a text editor to add or edit new entries thanks to how it stores its sensitive information.

(A caps lock option would be really helpful if this is implemented, though I don't know how simple that would be to implement in addition.)

_Bonus:_ in the same vain, a copy-paste option would be really good too. However I don't expect that to be as simple to implement as a general keyboard passthrough.

akopachov, 2023-01-30:
Not directly related to the topic, but still I'd like to explain as an author of Flipper Authenticator:

Usage of GUI to add new tokens was never the best option. That GUI was added at POC phase and it meant to be used when there is no chance to use other options (which is rather rare case).

First of all, there is CLI command `totp` which allows to do lot of stuff configuring the app and manipulating tokens. You may get list of available commands and options by running `totp help`. One of the commands available is [`totp add`](https://github.com/akopachov/flipper-zero_authenticator/wiki/How-to-add-new-token%3F#using-cli) which allows to add new token in a much more friendly way (compared to GUI) including ability to copy&paste. Personally I think this is the most user-friendly option.

Also, there is an option to put plain unencrypted token secret straight into a config file. After that, once application will be launched it will notice that there is plain token in a config file, it will encrypt it and replace plain token with encrypted representation. So Flipper Authenticator always checks and cares to make sure everything what should be encrypted is actually encrypted.

All of these options are described [here in Flipper Authenticator Wiki](https://github.com/akopachov/flipper-zero_authenticator/wiki/How-to-add-new-token%3F)

Feel free to raise a discussion in Flipper Authenticator [discussion section](https://github.com/akopachov/flipper-zero_authenticator/discussions) if you have any other questions or ideas.
Also keep in mind that I'm still in progress of writing wiki, so be patient if you see "TBD" somewhere. Or even better - suggest an edit to fill it up. This will allow newbies to have much better places to get answers for typical questions.

gsurkov, 2023-01-30:
In short, this would require a separate protocol for transmitting keyboard input and adapting the existing entry widgets to make use of it. We have chosen not to implement it because the resource requirements and maintainability issues outweigh the benefits. So it's a no here, sorry. Please feel free to report any other bugs or post suggestions.

closed 2023-01-30 by gsurkov