flipperdevices/qFlipper #161: AppImage doesnt start in Ubuntu 22.04

bfgelectronics, 2023-01-30:
**Describe the bug**
I was trying to execute the app image in ubuntu.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the latest AppImage from the official flipper website, 1.2.2 was the version in my case
2. runned chmod + x on the AppImage
3. executed the AppImage
4. The App image crashed with the error:
`Cannot mix incompatible Qt library (5.15.2) with this library (5.15.3)
/tmp/.mount_qFlippTNPrEz/AppRun.wrapped: line 87: 27249 Aborted                 (core dumped) "$THIS_DIR"/usr/bin/qFlipper`

**Expected behavior**
The AppImage should execute and open sucesfully.

**Actual behavior**
The app just crashed.

**Additional context**
[image]

gsurkov, 2023-01-31:
I was not able to reproduce this on an Ubuntu 22.04 VM.
Could you please give me the output of `apt list --installed | grep qt5`? This might help finding the package which causes the trouble.

bfgelectronics, 2023-02-01:
@gsurkov , this is the output of that command:
`base) bfg@bfg-System-Product-Name:~/freelance/eggAnalytics$ apt list --installed | grep qt

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libqt5concurrent5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5core5a/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5dbus5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5gui5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5network5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5sql5-sqlite/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5sql5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5svg5/jammy,now 5.15.3-1 amd64 [installed,automatic]
libqt5widgets5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5x11extras5/jammy,now 5.15.3-1 amd64 [installed,automatic]
libqt5xml5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
qt5-gtk-platformtheme/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
qttranslations5-l10n/jammy,jammy,now 5.15.3-1 all [installed,automatic]
vlc-plugin-qt/jammy,now 3.0.16-1build7 amd64 [installed,automatic]
`

bfgelectronics, 2023-02-01:
@gsurkov  this is the output of that comand:

```
base) bfg@bfg-System-Product-Name:~$ apt list --installed | grep qt

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libqt5concurrent5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5core5a/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5dbus5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5gui5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5network5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5sql5-sqlite/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5sql5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5svg5/jammy,now 5.15.3-1 amd64 [installed,automatic]
libqt5widgets5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
libqt5x11extras5/jammy,now 5.15.3-1 amd64 [installed,automatic]
libqt5xml5/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
qt5-gtk-platformtheme/jammy-updates,now 5.15.3+dfsg-2ubuntu0.2 amd64 [installed,automatic]
qttranslations5-l10n/jammy,jammy,now 5.15.3-1 all [installed,automatic]
vlc-plugin-qt/jammy,now 3.0.16-1build7 amd64 [installed,automatic]
```

gsurkov, 2023-02-01:
Installed all of the above packages, but was unable to reproduce the problem still.

Do you have any special Qt-related environment variables set? (You can find it out by running `env | grep QT`). A custom Qt style, maybe?

bfgelectronics, 2023-02-01:
This is the output on the terminal if i run the command you mentioned:
```
QT_ACCESSIBILITY=1
QT_IM_MODULE=ibus
```

gsurkov, 2023-02-01:
It's unlikely that the drivers cause the issue. (Unless they somehow messed with LD_LIBRARY_PATH or LD_PRELOAD).

Could you post the full output of the `env` command? (environment variables can contain personal data, check it before posting).

bfgelectronics, 2023-02-02:
This is the output of the env command, i looked over it and i dont see anything suspicious and also it shouldnt have any sensitive info:
```
SHELL=/bin/bash
SESSION_MANAGER=local/bfg-System-Product-Name:@/tmp/.ICE-unix/2276,unix/bfg-System-Product-Name:/tmp/.ICE-unix/2276
QT_ACCESSIBILITY=1
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
CONDA_EXE=/home/bfg/miniconda3/bin/conda
_CE_M=
LC_ADDRESS=ro_RO.UTF-8
GNOME_SHELL_SESSION_MODE=ubuntu
LC_NAME=ro_RO.UTF-8
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
LC_MONETARY=ro_RO.UTF-8
GTK_MODULES=gail:atk-bridge
PWD=/home/bfg
LOGNAME=bfg
XDG_SESSION_DESKTOP=ubuntu
XDG_SESSION_TYPE=x11
CONDA_PREFIX=/home/bfg/miniconda3
GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1
SYSTEMD_EXEC_PID=2295
XAUTHORITY=/run/user/1000/gdm/Xauthority
WINDOWPATH=2
HOME=/home/bfg
USERNAME=bfg
LC_PAPER=ro_RO.UTF-8
LANG=en_US.UTF-8
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:
XDG_CURRENT_DESKTOP=ubuntu:GNOME
VTE_VERSION=6800
CONDA_PROMPT_MODIFIER=(base) 
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/9c73e978_9f8c_482d_85f0_7e20d23f552d
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
TERM=xterm-256color
LC_IDENTIFICATION=ro_RO.UTF-8
_CE_CONDA=
LESSOPEN=| /usr/bin/lesspipe %s
USER=bfg
GNOME_TERMINAL_SERVICE=:1.112
CONDA_SHLVL=1
DISPLAY=:1
SHLVL=1
LC_TELEPHONE=ro_RO.UTF-8
QT_IM_MODULE=ibus
LC_MEASUREMENT=ro_RO.UTF-8
CONDA_PYTHON_EXE=/home/bfg/miniconda3/bin/python
LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu:/usr/lib
XDG_RUNTIME_DIR=/run/user/1000
CONDA_DEFAULT_ENV=base
LC_TIME=ro_RO.UTF-8
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
PATH=/home/bfg/miniconda3/bin:/home/bfg/miniconda3/condabin:/home/bfg/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin
GDMSESSION=ubuntu
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
LC_NUMERIC=ro_RO.UTF-8
_=/usr/bin/env
```

gsurkov, 2023-02-02:
I see that `LD_LIBRARY_PATH` is not empty. Try unsetting it before launching qFlipper AppImage: `LD_LIBRARY_PATH= ./qFlipper.... etc`

The second possible source of the problem may be the Conda installation (it might include PyQt, which could theoretically somehow override some of the bundled libraries). Try launching qFlipper in a clean shell with no Conda-related stuff loaded.

bfgelectronics, 2023-02-02:
It seems that the LD_LIBRARY_PATH is the problem and not the conda env. by running the qFlipper AppImage with this command  `LD_LIBRARY_PATH=  ./qFlipper-x86_64-1.2.2.AppImage` , it works (or at least i am on the connect flipper screen)

closed 2023-02-02 by bfgelectronics

gsurkov, 2023-02-02:
I will think about force-unsetting `LD_LIBRARY_PATH` in the AppImage environment so that there will be no problems in case it has been set manually.