flipperdevices/qFlipper #163: Problems with screen dpi/Windows

eried, 2023-02-10:
**Describe the bug**
The window does not resize correctly if you move between different dpi in multimonitor setups

**To Reproduce**
Steps to reproduce the behavior:
1. Open qFlipper, move window to a screen with lower scale than another, i.e. 100%
[image]

2. Close and reopen qFlipper, so the window position is remembered on that screen
3. Move to higher scale screen, i.e. 150%
4. Contents of the window will be truncated (it resizes the content, but the window frame is too small):
[image]

**Expected behavior**
Normal window resize

**Actual behavior**
Wrong resize

**Additional context**
Running on Windows 10 22H2

gsurkov, 2023-03-09:
@eried please try the development version to see if this was fixed.
The scaling behaviour was also changed (now qFlipper will switch to 2x scale at 175% system scale and not 150% as before).

eried, 2023-03-09:
sure!

> development version

no dev releases with binaries? 🤔

gsurkov, 2023-03-09:
> no dev releases with binaries? thinking
[Here](https://update.flipperzero.one/builds/qFlipper/dev/), as usual!

eried, 2023-03-09:
Works perfect on the same setup that the old version produced this dpi error, nice!

closed 2023-03-09 by eried