flipperdevices/qFlipper #168: macOS 13.2 SIGSEGV when updating firmware from file.

lvfrazao, 2023-03-15:
While trying to install custom firmware from a file the qFlipper application crashed with a seg fault.

```
Crashed Thread:        12  Thread (pooled)

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000038
Exception Codes:       0x0000000000000001, 0x0000000000000038

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [31384]
```

Stack trace:
```
Thread 12 Crashed:: Thread (pooled)
0   qFlipper                      	       0x104835dcc TarArchive::root() const + 0
1   qFlipper                      	       0x104837810 TarZipUncompressor::extractFiles() + 40
2   qFlipper                      	       0x104837d5c QtConcurrent::RunFunctionTaskBase<void>::run() + 80
3   qFlipper                      	       0x105e27748 QThreadPoolThread::run() + 220
4   qFlipper                      	       0x105e23680 QThreadPrivate::start(void*) + 332
5   libsystem_pthread.dylib       	       0x1a0dbe06c _pthread_start + 148
6   libsystem_pthread.dylib       	       0x1a0db8e2c thread_start + 8

Thread 12 crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000000   x1: 0x00000001048377e8   x2: 0x0000000128a08200   x3: 0x000000000000000b
    x4: 0x0000000000000000   x5: 0x00000000000000a0   x6: 0x0000000000001000   x7: 0x000000003b9ac860
    x8: 0x0000000000000000   x9: 0x0000600001c37080  x10: 0x0000000000bf3d01  x11: 0x00000000005f8c00
   x12: 0x0000000128800028  x13: 0x00000000ffff800b  x14: 0x0000000000001000  x15: 0x00000000ffff7dff
   x16: 0x00000001288081f8  x17: 0x00000001a0c22640  x18: 0x0000000000000000  x19: 0x0000600001c37080
   x20: 0x0000600001c37480  x21: 0x00006000006f0040  x22: 0x0000600001c37480  x23: 0x0000000000000001
   x24: 0x57b419022e2f7e49  x25: 0x0000000000000090  x26: 0x0000000000000001  x27: 0x000000010df6bae0
   x28: 0x0000000000000000   fp: 0x000000016b8a6e90   lr: 0x0000000104837810
    sp: 0x000000016b8a6e30   pc: 0x0000000104835dcc cpsr: 0x60001000
   far: 0x0000000000000038  esr: 0x92000006 (Data Abort) byte read Translation fault
```

**To Reproduce**
Steps to reproduce the behavior:
1. Download unleashed firmware 0.34 from here: https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-034
2. Open qFlipper application
3. Click on 'Install from File'
4. Select `~/Downloads/flipper-z-f7-update-unlshd-034-2.tgz`

**Expected behavior**
Device is updated with new firmware, qFlipper does not crash.

**Actual behavior**
qFlipper crashes, device is not updated to new firmware.

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? **YES**
2. Does reconnecting/rebooting Flipper solve it? **NO**

[qFlipper-20230314-204255.txt](https://github.com/flipperdevices/qFlipper/files/10975797/qFlipper-20230314-204255.txt)

gsurkov, 2023-03-15:
Does it crash if you try to install the official firmware from [here](https://update.flipperzero.one/builds/firmware/release/flipper-z-f7-update-release-13032023-b698126c.tgz)?

jroberts235, 2023-03-23:
I was able to install the official firmware, but both qFlipper and the iPhone app crash when attempting to install the unleashed-firmware, as described above.

jroberts235, 2023-03-23:
I found that the shasum was not correct after I downloaded via Safari (OSX was trying to un-compress after download and had errors).  I got the firmware installed by pulling the file down using wget instead and had no problems installing via qFlipper.

gsurkov, 2023-03-24:
@jroberts235 Does qFlipper crash if you use the file downloaded by Safari?