flipperdevices/qFlipper #175: no more drag and drop?

KasoudraKZK, 2023-03-30:
**in an old version of Qflipper (dating from 1-2 months ago) it was possible to drag & drop files from windows to the SDcard via Qflipper, it was much more practical to move folders with subfolders without going through an adapter for the SDcard. this function has disappeared since the last 2 or 3 updates.**

I tried to reinstall, update but nothing works and 2 of my friends have the same issue which must be the same for everyone 

_*there's nothing at all in the logs that can be used for this problem*_

gsurkov, 2023-03-30:
What operating system are you using?
Could you go through [the old versions](https://update.flipperzero.one/builds/qFlipper/) and find the last one which is working for you?

KasoudraKZK, 2023-03-30:
I tried V 1.3.0 but it doesn't work I know it wasn't a drag and drop bug,
I don't know which version works, I'll test it later

I'm on Win10 pro 21H2

gsurkov, 2023-03-30:
@KasoudraKZK are you running qFlipper with administrator privileges?

KasoudraKZK, 2023-03-31:
yes indeed that was it...I did not know this rule..thank you

 for those who have the same problem check in the properties>compatibility of 
Qflipper that the box ''run this program as admin'' is disabled

closed 2023-03-31 by KasoudraKZK