flipperdevices/qFlipper #178: qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key

wsy2220, 2023-04-06:
**Describe the bug**
Remote control using on screen keys not working
**To Reproduce**
1. Open remote control screen
2. Press any on screen keys

**Expected behavior**
Flipper zero get key input

**Actual behavior**
Nothing happens, with error in log:
```
qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
```
**Additional troubleshooting**
1. version 1.2.2 works

**Additional context**
```
158 [APP] qFlipper version 1.3.0 commit 01fcfec9 2023-03-27T17:51:38
158 [APP] OS info: Debian GNU/Linux 12 (bookworm) 12 6.2.2 Qt 5.15.8
172 [REG] Detected new device: VID_0x483:PID_0x5740
339 [DBG] Trying serial port  at /dev/ttyUSB0
339 [DBG] Trying serial port flip_******* at /dev/ttyACM0
339 [DBG] Using  serial port flip_******* at /dev/ttyACM0
339 [RPC] Starting RPC session...
438 [RPC] RPC session started successfully.
438 [RPC] (1) System Protobuf Version START
501 [DBG] Detected protobuf version: 0.15
501 [RPC] (1) System Protobuf Version SUCCESS
501 [RPC] (2) Property Get START
584 [UPD] Fetched update information from https://update.flipperzero.one/firmware/directory.json
617 [RPC] (2) Property Get SUCCESS
618 [RPC] (3) Storage Info @/ext START
669 [RPC] (3) Storage Info @/ext SUCCESS
669 [RPC] (4) Storage Stat @/ext/Manifest START
731 [RPC] (4) Storage Stat @/ext/Manifest SUCCESS
734 [RPC] (5) System Get DateTime START
786 [DBG] Flipper time skew is -609 milliseconds
786 [RPC] (5) System Get DateTime SUCCESS
786 [RPC] (6) System Set DateTime START
838 [RPC] (6) System Set DateTime SUCCESS
838 [RPC] Stopping RPC session...
840 [RPC] Attempting to unload protobuf plugin...
840 [RPC] Unloaded protobuf plugin.
840 [RPC] RPC session stopped successfully.
840 [REG] Registering the device
840 [DEV] Version: 0.80.1 commit: c41fe8e4 radio: 1.13.3
840 [BKD] Current device changed to "*******"
951 [RPC] Starting RPC session...
1105 [RPC] RPC session started successfully.
1105 [RPC] (1) Gui Start ScreenStream START
1168 [RPC] (1) Gui Start ScreenStream SUCCESS
5003 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
5069 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
5086 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6070 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6137 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6153 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6453 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6530 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6536 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6775 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6853 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
6870 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
7571 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
7653 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key
7670 [default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key

```

gsurkov, 2023-04-06:
Looks like a Qt5-specific problem.
You can use the AppImage version as a workaround (it is compiled with Qt6) until the fix will be released.

wsy2220, 2023-04-06:
The AppImage version does work. Thanks.

gsurkov, 2023-04-10:
Fixed in 7a6770b. Will try to make a new release as soon as possible. 
However, the Debian package will have to be updated by its respective maintainers.

closed 2023-04-10 by gsurkov