flipperdevices/qFlipper #180: After taking screenshot display stops updating

mumixam, 2023-04-09:
after taking a screenshot from qflipper the screen stops updating.

using the binary image on Arch linux

sha1sum
4178c2edcce60ec6a1a5f27f1f503bc37359d0e6  qFlipper-x86_64-1.3.0.AppImage

gsurkov, 2023-04-09:
Fixed in 5d2603c.

closed 2023-04-09 by gsurkov

mumixam, 2023-04-09:
Great. I've confirmed this works but this fix introduced another issue. Now the navigation on screen button do not work. When you press them you get the following error.

```[default] qrc:/components/StreamOverlay.qml:75: Error: Unknown method parameter type: InputEvent::Key```

gsurkov, 2023-04-10:
It was already reported in #178. 
In any case, please create a separate issue for each bugs you are encountering. It's easy to overlook stuff in the comments.