flipperdevices/qFlipper #182: Shared Library libwayland-egl.so.1 Missing On Gentoo

CyberLeo, 2023-04-24:
First and foremost, thank you for your interest in making qFlipper better by reporting bugs!

**Describe the bug**
A freshly downloaded copy of qFlipper-x86_64-1.3.0.AppImage fails to run. The previous version 1.2.2 runs fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Download qFlipper-x86_64-1.3.0.AppImage
2. chmod u+x qFlipper-x86_64-1.3.0.AppImage
3. ./qFlipper-x86_64-1.3.0.AppImage
4. Error!

**Expected behavior**
qFlipper should launch and show its interface.

**Actual behavior**
The following error message appears in the terminal and the program ends:
`/tmp/.private/cyberleo/.mount_qFlippVUDmNB/usr/bin/qFlipper: error while loading shared libraries: libwayland-egl.so.1: cannot open shared object file: No such file or directory`

**Additional troubleshooting**
1. Does the issue persist if you try multiple times? Yes
2. Does reconnecting/rebooting Flipper solve it? No

**The report *must* also include at least one of the following:**
1. A **FULL LOG** of this application run. (see below).
2. A screenshot of the application window (if it is a graphical issue).

qFlipper never starts, so there are no logs nor windows generated.

**Additional context**
Gentoo GNU/Linux profile amd64/17.1, OpenRC, Xorg.

gsurkov, 2023-04-24:
If you are really running Xorg, then qFlipper should not need this library. Otherwise it must be installed in the system (via the package manager).

Could you please try running it with this command: 
```
./qFlipper-x86_64-1.3.0.AppImage gui -platform xcb
```
and see if it does start?

CyberLeo, 2023-04-24:
I am definitely not running wayland, and have no wayland libraries installed.

It looks like the dynamic linker is considering libwayland-egl.so.1 a mandatory dependency for usr/bin/qFlipper and is bailing out when it cannot be found.

Perhaps there's a way to make it an optional or runtime dependency at build time instead?

```
(e7f194dd)cyberleo@arba ~/download $ ./qFlipper-x86_64-1.3.0.AppImage gui -platform xcb
/tmp/.private/cyberleo/.mount_qFlippTj6CDP/usr/bin/qFlipper: error while loading shared libraries: libwayland-egl.so.1: cannot open shared object file: No such file or directory
```
No change.

gsurkov, 2023-04-25:
I've probably got an idea of what happened.
The latest release is the first one to support wayland in the AppImage version, and to avoid the problems coming from distributing the necessary platform plugins, we packaged qFlipper as a statically linked application. Which, in hindsight, made both Xcb and Wayland hard dependencies. It's rather surprising there were no bug reports about it all this time.

I'll have to think whether to revert to dynamic linking with Qt or to employ some trickery to make this work without redundant system libraries.